waveshareteam/e-Paper #2: Add to documentation that font needs to be installed on RPi

cryptax, 2019-03-28:
On a Raspberry Pi, to use the python3 examples, you also need to install `sudo apt-get install fonts-wqy-microhei` or you will get an error:

```
Drawing
traceback.format_exc():
%s Traceback (most recent call last):
  File "main.py", line 22, in <module>
    font24 = ImageFont.truetype('/usr/share/fonts/truetype/wqy/wqy-microhei.ttc', 24)
  File "/usr/local/lib/python3.5/dist-packages/PIL/ImageFont.py", line 280, in truetype
    return FreeTypeFont(font, size, index, encoding, layout_engine)
  File "/usr/local/lib/python3.5/dist-packages/PIL/ImageFont.py", line 145, in __init__
    layout_engine=layout_engine)
OSError: cannot open resource
```

This should be added to the documentation.

waveshare, 2019-03-29:
yes,thinks ,i will fix readme

k4m454k, 2019-03-31:
i found a font here.
https://github.com/byrongibson/fonts/tree/master/truetype/wqy

waveshare, 2019-04-01:
try:
sudo apt-get install ttf-wqy-zenhei ttf-wqy-microhei

waveshare, 2019-04-01:
Now you can see it in the readme.txt

closed 2019-04-01 by waveshare