waveshareteam/e-Paper #5: [7.5inch_b&c] Python File not found

narolinus, 2019-04-01:
I am using the 7.5 inch red/black Display with a Raspberry Pi Zero W running the latest Raspbian Stretch. I followed the Manual and while the BCM-C-Example is running nicely, the Python-Code (tried both Python 2 and 3) fails with the following error:

`Traceback (most recent call last):
  File "main.py", line 4, in <module>
    import epd7in5b
  File "/home/pi/e-Paper/7.5inch_e-paper_b&c_code/python3/epd7in5b.py", line 51, in <module>
    import epdconfig
  File "/home/pi/e-Paper/7.5inch_e-paper_b&c_code/python3/epdconfig.py", line 48, in <module>
    SPI = spidev.SpiDev(0, 0)
FileNotFoundError: [Errno 2] No such file or directory
`
Any suggestions?

miikeat, 2019-04-03:
The error message seems to come from a missing spi device.

If you also don't have any device starting with spi in ```/dev/``` run

```
sudo rpi-update
``` 
that fixed the issue for me after a reboot.

and also check if spi is enabled in ```config.txt```:

```
dtparam=spi=on
```

narolinus, 2019-04-07:
That fixes the problem. Thanks.

miikeat, 2019-04-07:
@narolinus I'd say leave that issue open, because this should be added to the documentation! And someone of waveshare should look into that issue so other dont have the same issue!

narolinus, 2019-04-07:
Well... you are right.

waveshare, 2019-04-08:
We will solve this problem

waveshare, 2019-04-08:
now you can see it in readme.txt

closed 2019-04-08 by waveshare