waveshareteam/e-Paper #9: epd7in5 not working

dabch, 2019-04-10:
In the current master branch, both the python and python3 demo codes for the black and white 7.5" model don't work. I haven't tried the bcm version yet, but with wiringPi everything works as intended. 
I have all python libraries installed (spidev and RPi.GPIO with pip3, python3-pillow with apt because the pip installation failed) and there are no error messages printed by the program. Instead, main.py seems to run just fine - the only problem is that the display shows no reaction.

I went back to commit dd0f1e8 and tried it again. Now the python code works, so something between then and HEAD broke it.

dabch, 2019-04-10:
Commit 96f7f57 seems to have broken it, one commit before (773eba0) the python demo code still works.

waveshare, 2019-04-11:
sorry,it look SendCommand and SendData error

waveshare, 2019-04-11:
now ,fix it,you can try it again

closed 2019-04-11 by waveshare