waveshareteam/e-Paper #11: Incorrect rendering of images that have been drawn over

dabch, 2019-04-11:
I am using Pillow to generate an image that is shown on the 7.5" black and white display. Therefore, I have to clear some areas that have already been drawn on by drawing a white rectangle over them: `d.rectangle((100, 100, 300, 150), fill=1)`
But when the image is shown on the screen, the area that I drew white over is drawn black. This is not a problem with Pillow, as the generated image is correct, but with the representation of the pillow-generated image on the screen. Please see the files attached: one is the image generated by Pillow (as saved in the python code) and one is a photo of what the screen actually shows.
[image]
[image]

waveshare, 2019-04-13:
because d.rectangle（（100,100） ，300,150），fill = 1）
is fill a black rectangle

dabch, 2019-04-13:
That is wrong. 1 seems to be white and 0 seems to be black. The code I wrote is 
`d.rectangle((100, 100, 300, 150), fill=1)`
`d.line((100, 100, 300, 150), fill=0)`
And you can clearly see in the second picture that the rectangle is white (color 1) and the line across the rectangle is black (color 0)!

dabch, 2019-04-13:
Also, changing the rectangle's color to 0 doesn't change how it is displayed on the screen, but generates a black rectangle.

dabch, 2019-04-13:
I found the solution, though I'm unsure if it's a bug or intended behavior:
https://github.com/waveshare/e-Paper/blob/e945bfbbf36877b7b6d46d24608b67d4c3cb33bd/7.5inch_e-paper_code/RaspberryPi/python3/epd7in5.py#L183 and the following lines convert the Pillow image to a black-and-white array. For values under 64, the pixel is interpreted as black, under 192 is red (for 3-color ePaper) and over 192 is white. So it's sufficient to simply fill=200 the rectangle.

closed 2019-04-16 by waveshare