waveshareteam/e-Paper #12: Is it possible to draw an image over an existing one?

cryptax, 2019-04-11:
I would typically like to draw a _background_ image, and always keep it.
Then, I would like to draw **over** that image some text.

In the sample code that you provide, `epd.display(epd.getbuffer(HBlackimage), epd.getbuffer(HRedimage))` clears the screen and displays a new image. It calls in `epd2in7b.py` the command `DISPLAY_REFRESH`. I tried `PARTIAL_DISPLAY_REFRESH` instead without success.

Could you please explain how to draw one image over another one if that is possible? Thanks.

I am using a **3-color** Waveshare **2.7''** screen (B), on a **Raspberry Pi Zero W**, based on your **Python 3** example.

caemor, 2019-04-12:
partial refresh is just a faster refresh rate where the screen ink particles are not fully reset, but you get higher refresh rate.

I haven't used the 2.7" (b) myself yet but it looks like the python version isn't supporting the partial upload functions (e.g. the arduino has the following: https://github.com/waveshare/e-Paper/blob/e945bfbbf36877b7b6d46d24608b67d4c3cb33bd/2.7inch_e-paper_b_code/arduino/epd2in7b-demo/epd2in7b.h#L94).

waveshare, 2019-04-13:
You can only draw in ImageDraw.Draw(HBlackimage), so you only need to pass it once.

waveshare, 2019-04-13:
2.7-inch e-Paper screen does not support partial refresh, otherwise it will cause diaphragm damage

closed 2019-04-16 by waveshare

cryptax, 2019-04-20:
> You can only draw in ImageDraw.Draw(HBlackimage), so you only need to pass it once.

I'm not sure to understand. Do you mean I could chose for instance to draw in black a background image, only once, and then refresh another image or text in red over it?