waveshareteam/e-Paper #13: Is this buffer being created correctly?

kdrakon, 2019-04-12:
https://github.com/waveshare/e-Paper/blob/e945bfbbf36877b7b6d46d24608b67d4c3cb33bd/1.54inch_e-paper_code/RaspberryPi/python3/epd1in54.py#L200

The above is happening within a nested loop. As an example, in the one iteration, `y` could be `0` and `x` could be `1`. On a subsequent execution of the loop, `y` could increment to `1` and `x` restart to `0`. In both cases, the referenced buffer index would be the same. This would essentially happen numerous times. Maybe I'm missing something else, but that doesn't seem to be right. Is there any reason we need to put the image buffer data into a single dimension array? Why not a two-dim array, like the image data?

Also, while I'm here, I can't find any documentation that states why we need to take the floor division of 8 (`// 8`) for the height and width of the display. Can you point to where I can find this? I noticed the floor division varies for the displays.

caemor, 2019-04-12:
Your calculation is wrong. Assuming you are talking about the 1.54in display with a 200x200 resolution the calcuation is the following:
For y=0 && x=1:
(1 + 0 * 200) // 8 = 0
For y=1 && x=0: 
(0 + 1*200) //8 = 25

For the floor division:
Each Byte of the image buffer contains the data for 8 different pixels, therefore you divide the "pixel position" by eight to get the correct position in the array.

kdrakon, 2019-04-12:
Oh yeah, my mistake; wasn't following the order of operations correctly in my head. Thanks also for explaining the image buffer 👍.

closed 2019-04-12 by kdrakon