waveshareteam/e-Paper #15: display never reaches deep sleep

qrti, 2019-04-18:
## Arduino Library for Black and White 2.9 inch e-paper
displays of other sizes and platforms might be concerned as well

### original in epd2in9.cpp
```
void Epd::Sleep() {
    SendCommand(DEEP_SLEEP_MODE);
    WaitUntilIdle();
}
```
result

- the display never reaches deep sleep, because a following 0x01 data byte is missing
- WaitUntilIdle() returns because the display is not sleeping
- because the display never sleeps it might be damaged, as mentioned in the manual

### proposal

#### 1. change in epd2in9.cpp
```
void Epd::Sleep()
{
    // datasheet prescribes turning off
    // oscillator clock and DC/DC & regulator
    // befor entering sleep
    // but does not tell how to do it

    SendCommand(DEEP_SLEEP_MODE);
    SendData(0x01);                     // 0x01 = deep sleep mode

    pinMode(CS_PIN, INPUT);             // prevent parasitic supply
    pinMode(RST_PIN, INPUT);
    pinMode(DC_PIN, INPUT);

    SpiEnd();                           // optional
}
```

#### 2. add to epdif.cpp (optional)
```
void EpdIf::SpiEnd(void)
{
    SPI.end();
}
```
#### 3. add to epdif.h (optional)
```
static void SpiEnd(void);
```

I've done a lot of research, reading, testing, measuring to make these displays run as intended

waveshare, 2019-04-19:
sure, you are right

lochel, 2019-05-09:
Has it already been fixed?

cymerio, 2019-05-11:
I've been hit with this bug aswell (already damaged 3 displays!)
I've been digging for a solution and I've found this code somewhere in the net:
```python
    def power_on(self):
        self.send_command(DISPLAY_UPDATE_CONTROL_2) #0x22
        self.send_data(0xC0)
        self.send_command(MASTER_ACTIVATION) #0x20

    def power_off(self):
        self.send_command(DISPLAY_UPDATE_CONTROL_2) #0x22
        self.send_data(0xC3)
        self.send_command(MASTER_ACTIVATION) #0x20
```
I suspect we need to send the `power_off` command before `sleep`.
And indeed the [new demo code](https://www.waveshare.com/wiki/File:2.13inch_e-paper_code(V2).7z) (for displays purchased after 2019-04-26) has this:
```python
    def sleep(self):
        self.send_command(0x22) #POWER OFF
        self.send_data(0xC3)
        self.send_command(0x20)

        self.send_command(0x10) #enter deep sleep
        self.send_data(0x01)
        epdconfig.delay_ms(100)    
```
Turning output pins to input also look like a good idea. I have to try it.

Couple of questions:
1. Is there a known way to check **for sure** if the display is in deep sleep state? Do you simply measure current consumption at the display? Or is there a place where you can measure a voltage drop?
My display has these measuring points: TPA, TPB, TPC, TPD, TPF and CL, but I have no idea what they are. Measuring at the connector also seems tricky.
2. Since the displays were damaged by using bad code from the manufacturer, can we ask for refunds or replacements?

cymerio, 2019-05-12:
# My new findings
I measured current between the +3.3V pin of the Raspberry Pi 3B and the VCC pin of the 2.13" e-paper HAT (display dated 2017-06-02).

Given this code:
```python
def sleep_bad():
    send_command(0x10)
    wait_until_idle()
```

```python
def sleep_good():
    send_command(0x10)
    send_data(0x01)
    delay_ms(100)    
```

```python
def power_off():
    send_command(0x22)
    send_data(0xC3)
    send_command(0x20)
```

- Current after `init`: **495 μA**
- Current while sending an image: **3300 μA** (fluctuates up to 5400)
- Current after sending an image: **1110 μA**
- Current after `sleep_bad`: **629 μA** or the last reading if it was greater
  - *Example 1*: `init` (495) then `sleep_bad`: *629 μA*
  - *Example 2*: send image (1110) then `sleep_bad`: *1110 μA*
  - **Conclusion: `sleep_bad` doesn't work.**
- Current after `sleep_good`: **353 μA** (lowest)
- Current after `power_off`: **629 μA**
  - *Observation*: `power_off` then `sleep_good` makes no difference than just using `sleep_good`
  - **Conclusion: `power_off` is useless.** It's a bit better than doing nothing after sending an image (1110 μA decreases to 629 μA). But sleep is better (decreases to 353 μA).
- Current after `sleep_good`, then setting OUTPUTS as INPUTS (`SPI.close()`): **376 μA** (greater than just `sleep_good`). But since I couldn't measure the current at the 3 output pins I can't know for sure if total current increased or decreased. Notice I'm only measuring current through VCC.
- Current after `GPIO.cleanup()`: (no change)

## Final conclusions:
The code in `sleep_good` is the right one for my display (check size and date).
Nothing more is needed.

closed 2019-06-05 by waveshare

ubijdasilva, 2021-02-05:
Dear all,

I have seen your consumption report. I think 376µA is not the expected current for deep sleep mode :

[image]

Is everyone allowed to obtain this power consumption in deep sleep mode?

DarkZeros, 2021-12-17:
Hi
I have 3 diferent sets of displays. And can confirm something is going on there.

2x 154_D67 V1 -> 500uA after deep sleep
2x 154_D67 V2 -> 10uA after deep sleep
2x 154_Z90c V2 -> 500uA after deep sleep

Exactly same commands sent to display (power off then deep sleep)
And I can also confirm sending poweroff is not needed. It doesnt affect the deep sleep values.

Does anyone have any idea?

NishantEmbedded, 2023-08-04:
how to put the sleep mode in stm32,nRF52832 using of 2.9inch e-paper display?

NishantEmbedded, 2023-08-04:
I will also write the power off function in update time but it's still getting more current I don't know why the deep sleep mode command is not working.

DarkZeros, 2023-08-10:
Hi, I foundthe issue was related to the RES# pin, it needs to be pulled up for deepsleep to work properly.
https://github.com/sqfmi/watchy-hardware/issues/14