waveshareteam/e-Paper #17: Cannot wakeup from sleep

cryptax, 2019-04-24:
The epd2in7b library does not provide any function to wake up from `sleep()`. Once it is called, `reset()` for instance does not wake up the screen. Please assist to wake up the screen.
Should we send the command `POWER_ON`? and something else?

For example, see the program below. We can display text as much as we want until `epd.sleep()`. Once sleep is called, the program hangs during display of text number 3.

```python
#!/usr/bin/python
# -*- coding:utf-8 -*-

import epd2in7b
from PIL import Image,ImageDraw,ImageFont

print("Clear the screen")
epd = epd2in7b.EPD()
epd.init()
epd.Clear(0xFF) # clears the screen

def display_text(x,y, textblack, textred):

    HBlackimage = Image.new('1', (epd2in7b.EPD_HEIGHT, epd2in7b.EPD_WIDTH), 255) # 264 x 176 
    HRedimage = Image.new('1', (epd2in7b.EPD_HEIGHT, epd2in7b.EPD_WIDTH), 255)  
    drawblack = ImageDraw.Draw(HBlackimage)
    drawred = ImageDraw.Draw(HRedimage)
    
    font36 = ImageFont.truetype('/usr/share/fonts/truetype/wqy/wqy-microhei.ttc', 36)
    drawblack.text((x, y), textblack, font=font36)
    drawred.text((x,y+30), textred, font=font36)

    print("displaying...")
    epd.display(epd.getbuffer(HBlackimage), epd.getbuffer(HRedimage))

# test
print("Display 1: ")
display_text(30, 40, "Hello", "World")
print("Display 2: ")
display_text(30, 40, "Greetz", "@cryptax")

# sleep
epd.sleep()
print("Display 3: ")
display_text(30, 40, "Never", "see")
```

sspooner, 2019-04-24:
Can you epd.init() after your sleep ? I don't use python for my code, but in C I do EPD_Sleep() and then before I want to update the display I do an EPD_Init() again.  The time between my EPD_Sleep() and EPD_Init() is about 10 minutes for me.

cryptax, 2019-04-24:
Yes, `epd.init()` works.
It's just that I would have expected a lighter wake up. Okay, so let's do init.

closed 2019-04-24 by cryptax