waveshareteam/e-Paper #22: Black screen / no life

dicijr, 2019-06-11:
Similarly to #18 after maybe 2-3 weeks of using the e-Paper, my display has stopped being able to be updated. At first it was frozen, but now its a black screen with the frozen image very faint. I have tried running the demo code and it still doesn't clear it. Using RPI zero w 7.5in black (no red) display. Thanks.

hnwangkg-ezio, 2019-06-12:
How fast is your refresh rate, and your situation seems to have two problems:
1. It is refreshed and has not entered sleep mode.
2. Continuous refresh interval is less than 3 minutes

dicijr, 2019-06-12:
I'm a bit new to this stuff, I had used this project: https://github.com/mgfcf/E-Paper-Calendar & it was going fine until the other day, when it stopped refreshing. I assumed using the demo code would help me determine if it was hardware or software.

dicijr, 2019-06-12:
[image]

dicijr, 2019-06-12:
sudo python3 /home/pi/e-Paper/7.5inch_e-paper_code/RaspberryPi/python3/main.py:

e-Paper busy
e-Paper busy release
Clear
e-Paper busy

Is this what you mean its not in sleep mode?

dicijr, 2019-06-12:
Is it possible my unit is defective and I should try to get it replaced on ebay?

closed 2019-06-24 by waveshare