waveshareteam/e-Paper #23: clarify licensing

pabigot, 2019-07-02:
The licensing of the content of this repository is unclear; I"m specifically interested in the low-level epdXinY.* files.  The files themselves have an MIT license, but there's copies of GPL3 in `LICENSE` files in the parent directories.

Are these files still MIT licensed?  If so, why was the GPL3 license added?

hnwangkg-ezio, 2019-07-03:
What do you mean by that part of the code: Arduino or Stm32

pabigot, 2019-07-03:
[This file](https://github.com/waveshare/e-Paper/blob/master/Arduino%20UNO/epd1in54c/epd1in54c/epd1in54c.cpp) has the MIT license, as have all previous versions I've seen on the Waveshare wiki.

[This file](https://github.com/waveshare/e-Paper/blob/master/Arduino%20UNO/epd1in54c/LICENSE) is the GPL3 license.  I don't know what it's supposed to refer to.  These licenses files (in all the sibling directories) are the only material in the repository that appears to be associated with GPL.

Why is the GPL3 license in this repository?  What content in the repository is not MIT licensed?

hnwangkg-ezio, 2019-07-03:
MIT

closed 2019-07-04 by waveshare