waveshareteam/e-Paper #26: (De)Capitalization issues in the code

do5rsw, 2019-07-26:
This is an issue no bug or fault. Thus it is somehow annoying ;)
I assume you are developing on the windows platform which does not care for context sensitivity at all. So to you it don't matter if it reads #include <arduino.h> or #include <Arduino.h>.
On Unix/Linux and MacOS plattform it does matter. However context sensitivity is obviously preserved upon uploading your code to github. You see that in the code because in the sketch itself e.g. it reads #include <SPI.h> (correct) while in some header file it reads #include <spi.h>.
This will compile on windows with arduino ide but not on linux/unix/macos. 

You will have to correct the non-found headers here. The IDE will report them once you attempt to compile. It has to be #include <Arduino.h> and #include <SPI.h>. 
After correcting this it should compile.

For me that is not a problem for I am experienced with programm in (avr) c. But this can become really tough for unexperienced users.

hnwangkg-ezio, 2019-08-01:
Please try the program on github, don't use our English wiki, the program there has not been updated yet.

hnwangkg-ezio, 2019-08-01:
You are right, in fact, for most Arduino users, their platform is under windows, so there is no test to compile under linux environment.

closed 2019-08-12 by waveshare