waveshareteam/e-Paper #28: Python partial update on epd_2in9d damaging display?

errolt, 2019-07-30:
The python partial update code for epd_2in9d appears to be damaging my display. Instead of flipping changed pixles only(only set pixels that have changed), it updates all pixels, setting black pixels more black at every update, and every white pixel more white. This causes the image to burn into the display, making it hard to erase image.

errolt, 2019-07-30:
This code is in python code, but not in C code:
https://github.com/waveshare/e-Paper/blob/4e72f31f116d2ec2e9434884d6388792605cbe52/Python/lib/waveshare_epd/epd2in9d.py#L315-L318

errolt, 2019-07-30:
These lines also differ from the values used in C:
https://github.com/waveshare/e-Paper/blob/4e72f31f116d2ec2e9434884d6388792605cbe52/Python/lib/waveshare_epd/epd2in9d.py#L221-L224
https://github.com/waveshare/e-Paper/blob/4e72f31f116d2ec2e9434884d6388792605cbe52/Python/lib/waveshare_epd/epd2in9d.py#L243-L246

errolt, 2019-07-30:
After making these changes to match the C code, I get an inverted(black is white, white is black) image during partial update, but it no longer burns the image into the display. Image can erase after only one Clear command.

hnwangkg-ezio, 2019-08-01:
Trying to use our latest program, will there be such a problem?

closed 2019-08-12 by waveshare