waveshareteam/e-Paper #29: 7.5inch-e-paper demo / Debug: e-Paper busy

skgen, 2019-07-31:
Hi,

I'm trying to make the 7.5inch e-Paper working, I followed the tutorial on the documentation (https://www.waveshare.com/w/upload/7/74/7.5inch-e-paper-hat-user-manual-en.pdf) without any issue until running the programm.
Both python2 and C (wiringPi example) return same error about e-Paper being busy
Bcm2835 doesn't display any error but does'nt work neither.

`7.5inch e-Paper demo`
`set wiringPu lib success !!!`
`Debug: e-Paper busy`
`Debug: e-Paper busy release`
`clear...`
`Debug: e-Paper busy`
`Debug: e-Paper busy release`
`Paint_NewImage`
`WidthByte = 80, HeightByte = 384`
`EPD_WIDTH / 8 = 15`
`show bmp for window`
`pixel = 100 * 100`
`Debug: e-Paper busy`
`Debug: e-Paper busy release`

Im using RPI 3 b, raspbian stretch (latest), having same errors using the shield or the directly the wires.

If you have any idea why I do have this error you're welcome !
(If you need more informations, I'll try to answer you the best as possible)
Thanks !

hnwangkg-ezio, 2019-08-01:
Please try the program on github, don't use our English wiki, the program there has not been updated yet.

btutovic, 2019-08-01:
Those aren't errors, they're just debug statements. The function "ReadBusy" prints them whenever it's called:
```python
def ReadBusy(self):
  logging.debug("e-Paper busy")
  while(epdconfig.digital_read(self.busy_pin) == 0):      # 0: idle, 1: busy
    epdconfig.delay_ms(100)
  logging.debug("e-Paper busy release")
```

ozcam, 2019-08-11:
Same thing happens for me using github code of epd1in54. Nothing happens on the ink display just the debug info in a terminal window

closed 2019-09-02 by waveshare