waveshareteam/e-Paper #30: 1.54 DEBUG:root:e-Paper busy

ozcam, 2019-08-11:
When running this file -  epd_1in54c_test.py

The ink display screen does not change all I get is the debug readout in ssh terminal

INFO:root:epd1in54c Demo
INFO:root:init and Clear
DEBUG:root:e-Paper busy
DEBUG:root:e-Paper busy release
DEBUG:root:e-Paper busy
DEBUG:root:e-Paper busy release
INFO:root:1.Drawing on the image...
DEBUG:root:Horizontal
 Any help or suggestions

waveshare, 2019-08-12:
Are you sure the e-paper is not damaged?

closed 2019-09-02 by waveshare

adrianlol2017, 2020-05-16:
I have the same problem but I know for sure that my screen not is damaged

ragmondo, 2020-05-25:
Same here. How do they close the issue without offering a solution ? Brand new eink display purchased from amazon ... double checked the wiring and also the power to the pi. Removed any extraneous peripherals. Still doesn't work. Buyer beware

kevinetore, 2020-05-26:
@adrianlol2017 @ragmondo I'm getting the same issue, did you guys find a solution?

jpoles1, 2020-06-07:
I am having the same issue. My PiZero 2.13 in epaper HAT was working, I made some edits to my python code (changing what text was displayed, nothing major), it got stuck on busy, I reverted to the old code, still stuck. Can't get it back working again.

jpoles1, 2020-06-07:
I think it must just be an issue related to failure to receive the ready signal to end the busy state. I've circumvented it by simply adding a timeout to override the busy lock if it gets stuck for too long, which seems to work most of the time (modifying epd2in13bc.py for my device, but guessing it'll work for others):

def ReadBusy(self):
        logging.debug("e-Paper busy")
        self.send_command(0x71);
        iter = 0;
        while(epdconfig.digital_read(self.busy_pin) == 0 and iter < 100): 
            self.send_command(0x71);
            epdconfig.delay_ms(100)
            iter+=1
        logging.debug("e-Paper busy release")

deguilardi, 2020-06-23:
same here. this won't work and it looks like waveshare just don't believe the issue

jpoles1, 2020-06-23:
@deguilardi , did you try making my modification? It's not perfect by any means, but I've had it working on a functioning device for the past couple weeks. 

I agree though, I wish @waveshare would acknowledge that this is still causing problems and re-open the issue.

the-dino-enthusiast, 2020-08-07:
@waveshare I'm getting the same exact issue for my 7in5V2 display.

Calling ReadBusy()

```
    def ReadBusy(self):
        logging.debug("e-Paper busy")
        self.send_command(0x71)
        busy = epdconfig.digital_read(self.busy_pin)
        while(busy == 0):
            self.send_command(0x71)
            busy = epdconfig.digital_read(self.busy_pin)
        epdconfig.delay_ms(200)
```

This never ends. Rebooting raspberry pi zero does nothing.

I can't Clear() my screen since this check never passes and I'm afraid of burn-in due to this. 

@waveshare - Can you please help us out here?

the-dino-enthusiast, 2020-08-07:
@jpoles1 - Did you end up finding a solve? I added this artificial timeout, but all it does is protect me against the infinite loop. I can't even run Clear() command anymore and I have a old image on the e-ink for a few days now. :(

jpoles1, 2020-08-07:
@the-dino-enthusiast unfortunately no definitive solution, but I've got it working just fine with my workaround. You can see my full codebase for my project with this display here:
https://github.com/jpoles1/2.13in-epaper-weather

hakarune, 2020-09-15:
> @the-dino-enthusiast unfortunately no definitive solution, but I've got it working just fine with my workaround. You can see my full codebase for my project with this display here:
> https://github.com/jpoles1/2.13in-epaper-weather

Yours works, but none of the @waveshare ones do. What does yours do different so I can implement it on what I'm setting up?

jpoles1, 2020-09-15:
> I think it must just be an issue related to failure to receive the ready signal to end the busy state. I've circumvented it by simply adding a timeout to override the busy lock if it gets stuck for [...]

@hakarune I'm not sure exactly what you mean by your question, but ^this is the major change that I made to get things working properly in my repo.

hakarune, 2020-09-18:
> @hakarune I'm not sure exactly what you mean by your question, but ^this is the major change that I made to get things working properly in my repo.

This helped me get it mostly setup, thank you. Now the only issue I am having is trying to draw bmp/jpg/png on the display. I have tried a bunch of methods but everything fails horribly on this board.

hakarune, 2020-09-19:
> This helped me get it mostly setup, thank you. Now the only issue I am having is trying to draw bmp/jpg/png on the display. I have tried a bunch of methods but everything fails horribly on this board.

Finally got it all 100%. Thanks again

jpoles1, 2020-09-19:
@hakarune, glad to hear it! I would encourage you to share your working code, so the next unfortunate person to have to stumble upon this issue will have some more info to work off of!

Avdiel, 2021-04-08:
Solved for me by answer above.  Simply run another test.
For the 7.5 in, these tests works for me:
	epd_7in5b_HD_test.py
	epd_7in5_HD_test.py 

The V2 tests got stuck in "busy".

hubmartin, 2021-09-23:
Hi, I did some analysis and only on RPI Zero and only sometimes I got firmware stuck in `busy`.
However this solved it. Two issues are the Python is sometimes slow on Zero and when the hat is connected by 40 pin connector you send 5V to the RPI gpio.
https://www.martinhubacek.cz/blog/waveshare-epaper-hat-issues/

philippludwig, 2022-10-17:
> Are you sure the e-paper is not damaged?

@waveshare I have the same issue with a brand new e-Paper display. Does this mean I got a faulty unit?

dviner, 2023-01-03:
I have this same issue with brand new 7.5 in V2.  Running the sample code results in the "e-Paper busy" message, and it never progresses.  Nothing appear on the screen either.  I'm using a Raspberry Pi 3B+ running Bullseye.

@waveshare , any updates?

dhvandermeer, 2023-01-03:
Same here. After accidentally ripping my connector of the 7.5 inch V2 black/white screen, I bought a new one, but this time with a new connector type. The old version had this connector and worked flawlessly, as can be seen here in my [instructable](https://www.instructables.com/E-paper-Calendar-Raspberry-Pi-With-E-ink-Screen-an/).
[image]

Then I received a new screen with a new connector (other mark on the connector) plugged it in and nothing!! It stucked at busy. So I returned the screen and bought a new one from another store, but same problem. One thing that partially worked was the fix below, but it makes the refresh very, very slow and most frustrating: at the end of the refresh the screen completely fades out making it unreadable. I ended up buying the black/red screen: also same 'busy' problem, but the fix below worked for the black/red screen most times.
I used a raspberry pi 2+, 3 and also 3 different HAT's and using the wires... no difference; it just stucks on busy, very frustrating. Since the fix below does something I really think there is something wrong with the library and the newer versions of the V2 screen. 

> @waveshare I'm getting the same exact issue for my 7in5V2 display.
>
> Calling ReadBusy()
>
> ```
>     def ReadBusy(self):
>         logging.debug("e-Paper busy")
>         self.send_command(0x71)
>         busy [...]

dviner, 2023-01-04:
I looked through all the library code for python, and some of the code for C.  It seems that the readBusy function is a bit diverse in how it operates.

In some cases, it sleeps between loops, but not all cases.
The amount of time it sleeps varies from size to size.

In some cases, it sends a command in the loop, but not all.
The command is 0x71.

In some cases, the busy signal is 0, but in others it's 1.

I made a spreadsheet outlining all the differences in this function.  https://docs.google.com/spreadsheets/d/11S7DDWuIJ85-4fvlvpt37gkGPyAdy1uu0lcAeK4npTg/edit#gid=0 is that sheet.

My guess is that there's a clear intention to sleep between each loop.  The duration of the sleep doesn't seem important since it varies from 5 ms to 200 ms.

I'm honestly not sure what to think about the fact that the return value of the busy pin is sometimes 0 and sometime 1.  That feels really odd to me.

It also seems odd that only some boards have the 0x71 command sent to the board in between reading the busy pin.

epenguins, 2023-03-04:
I have had the same issue with the `epd_7in5b_V2_test.py` example for several weeks.

Compiling and running the C example followed by a reboot fixed the issue for me, now the  `epd_7in5b_V2_test.py` example works. I have an RPi 3B+ on Raspbian Bullseye.

knallfr0sch, 2024-10-25:
> I think it must just be an issue related to failure to receive the ready signal to end the busy state. I've circumvented it by simply adding a timeout to override the busy lock if it gets stuck for too long

The [12.48 inch three-colour module](https://www.waveshare.com/wiki/12.48inch_e-Paper_Module#Raspberry_Pi) also works with this trick and the official test program `epd12in48b` (b for three-colour: black, white, red.)

I use the 40pin RaspberryPi connector to drive the display using a Raspberry Pi 3 Model B Plus Rev 1.3.

I also had to replace the Python `time.clock()` calls [with `time.perf_counter()` calls.](https://stackoverflow.com/questions/58569361/attributeerror-module-time-has-no-attribute-clock-in-python-3-8)

Also I had to change the pip installation scripts, since the newer Raspbian OS versions (bullseye?) do not allow installing python modules system-wide.

I think the whole wiki, scripts and documentation needs an update.

ldicarlo, 2025-09-05:
Hello @knallfr0sch , would you mind sharing your final code, because 
```python
    def M1_ReadBusy(self): # (4 times that of course, since 4 screens)
        self.M1_SendCommand(0x71)
        busy = epdconfig.digital_read(self.EPD_M1_BUSY_PIN)
        busy = not(busy & 0x01)
        iter = 0
        while(busy and iter <= 100):
            self.M1_SendCommand(0x71)
            busy = epdconfig.digital_read(self.EPD_M1_BUSY_PIN)
            busy = not(busy & 0x01)
            iter += 1
        time.sleep(0.2)
```
is apparently not enough, either in V1 or V2. 
Thanks