waveshareteam/e-Paper #31: Blinking screen - 2.9inch e-Paper Module (B)

tbitowt, 2019-08-13:
Hello,

I have problem with running 2.9 inch 3-color screen with Module. The problem is that the screen is blinking like on following video: https://youtu.be/-NTjjIM-cio . For this demo I have used ```epd2in9bc``` example.

I have written my own ESP8266 library and debugged it with Logic Analyzer and everything seems to be fine according to the diagram from datasheet (https://www.waveshare.com/w/upload/b/bb/2.9inch-e-paper-b-specification.pdf , diagram `BWR mode & LUT from OTP`)

Is it a common problem? I couldn't find anything on the internet. 

P.S. With one of the display struct from `GxEPD2` library (`GxGDEW029T5`) the screen is not blinking, but this seems to not implement 3-color printing.

lakeuk, 2019-08-13:
Isn't that a normal function of the colour epapers when doing their processing, I see this on all the demo videos

https://www.youtube.com/watch?v=RoYZxaUZues

btutovic, 2019-08-14:
@tbitowt, @lakeuk is correct, this is part of the screen update process. Sometimes referred to as "flashing" the image onto the screen. Nothing to worry about here as long as the screen is displaying the correct image.

btutovic, 2019-08-14:
Also be weary of updating the screen directly, this is called a "partial refresh" and I'd definitely consider it an advanced feature. If used incorrectly it will permanently burn the image into your e-paper screen. A full refresh takes longer, but prolongs the life of your screen. To be extra safe I like to Clear the screen before writing a new image to it.

GDChrisWu, 2019-08-23:
That is normal, now all three color EPD need a long time refresh time with flicker several time to clean the screen to avoid ghost, if you do not like this then you can try to use black & white color verion, the refresh time will be shorter and support partial refresh without flicker, but I suggest after 5 times partial refresh then add a full screen refresh to avoid the ghost

closed 2019-09-02 by waveshare