waveshareteam/e-Paper #32: How wake epaper display from sleep - using example Python script

lakeuk, 2019-08-15:
Using the test python3 script for 2in9, how do you wake up the display after a sleep within the same script. 

I've tried the initialise statement
`logging.info("Goto Sleep...")`
`epd.sleep()`
`epd.init(epd.lut_full_update)`  but it returns this error 

>     epd.init(epd.lut_full_update)
>   File "/home/myapp/epaper_python3/epd2in9.py", line 118, in init
>     if (epdconfig.module_init() != 0):
>   File "/home/myapp/epaper_python3/epdconfig.py", line 71, in module_init
>     self.SPI.max_speed_hz = 4000000
> OSError: [Errno 9] Bad file descriptor

Script - e-Paper/RaspberryPi&JetsonNano/python/examples/epd_2in9_test.py

hnwangkg-ezio, 2019-08-20:
becaus epdconfig.py have：
`def module_exit(self):
        logging.debug("spi end")
        self.SPI.close()`
change:
`def module_exit(self):
        logging.debug("spi end")
        # self.SPI.close()`

lakeuk, 2019-08-21:
Thanks, worked for me, much appreciated

closed 2019-08-21 by lakeuk