waveshareteam/e-Paper #37: LUT color modification

AlexisSADAL, 2019-10-10:
Hi,

I tried to modify the LUT initilisation by modifying values inside them. 
But my modification doesn't changed anything on the refresh process,
I am pretty sure that the COG keep is previous LUT in memory and doen't take the new one into account. 
I am working on 1inch54 3 colors (B, W, Y) and I would like to optimize the transition time even if some ghosts keep on the screen.
Is anybody try to modify LUT this screen or another ? 

Alexis

caemor, 2019-10-11:
You just need to write to the LUT_Registers... 
I haven't tried any of the colored ones yet, but for the 1.54in BW one I have some code here:   https://github.com/caemor/epd-waveshare/blob/master/src/epd1in54/mod.rs#L353-L361

So you need to send a Command to the WRITE_LUT_REGISTER and then send the data (your new lut values...)

AlexisSADAL, 2019-10-14:
Yes that is I am doing.
I send the command and after that the data. 
And the refresh process stay the same even if I change values in my LUT data.

waveshare, 2019-10-18:
no, you cann't

closed 2019-10-22 by waveshare