waveshareteam/e-Paper #41: Setting partials not working properly?

randohinn, 2019-11-04:
I'm trying to adapt this to a pure ATMEGA48 and having some trouble setting the image partial.. Here's my implementation so far

```
#include "spi.h"
#include <util/delay.h>

// Display resolution
#define EPD_WIDTH       104
#define EPD_HEIGHT      212

#define IMAGE_WIDTH 16
#define IMAGE_HEIGHT 16
#define IF_INVERT_COLOR 0

// EPD2IN13B commands
#define PANEL_SETTING                               0x00
#define POWER_SETTING                               0x01
#define POWER_OFF                                   0x02
#define POWER_OFF_SEQUENCE_SETTING                  0x03
#define POWER_ON                                    0x04
#define POWER_ON_MEASURE                            0x05
#define BOOSTER_SOFT_START                          0x06
#define DEEP_SLEEP                                  0x07
#define DATA_START_TRANSMISSION_1                   0x10
#define DATA_STOP                                   0x11
#define DISPLAY_REFRESH                             0x12
#define DATA_START_TRANSMISSION_2                   0x13
#define VCOM_LUT                                    0x20
#define W2W_LUT                                     0x21
#define B2W_LUT                                     0x22
#define W2B_LUT                                     0x23
#define B2B_LUT                                     0x24
#define PLL_CONTROL                                 0x30
#define TEMPERATURE_SENSOR_CALIBRATION              0x40
#define TEMPERATURE_SENSOR_SELECTION                0x41
#define TEMPERATURE_SENSOR_WRITE                    0x42
#define TEMPERATURE_SENSOR_READ                     0x43
#define VCOM_AND_DATA_INTERVAL_SETTING              0x50
#define LOW_POWER_DETECTION                         0x51
#define TCON_SETTING                                0x60
#define RESOLUTION_SETTING                          0x61
#define GET_STATUS                                  0x71
#define AUTO_MEASURE_VCOM                           0x80
#define READ_VCOM_VALUE                             0x81
#define VCM_DC_SETTING                              0x82
#define PARTIAL_WINDOW                              0x90
#define PARTIAL_IN                                  0x91
#define PARTIAL_OUT                                 0x92
#define PROGRAM_MODE                                0xA0
#define ACTIVE_PROGRAM                              0xA1
#define READ_OTP_DATA                               0xA2
#define POWER_SAVING                                0xE3
/*
unsigned char epaper_get_image(int width) {
	return malloc(8*width);
}*/

void epaper_draw_absolute_pixel(unsigned char* image, int x, int y, int colored) {
	if (x < 0 || x >= IMAGE_WIDTH || y < 0 || y >= IMAGE_HEIGHT) {
        return;
    }
    if (IF_INVERT_COLOR) {
        if (colored) {
            image[(x + y * IMAGE_WIDTH) / 8] |= 0x80 >> (x % 8);
        } else {
            image[(x + y * IMAGE_WIDTH) / 8] &= ~(0x80 >> (x % 8));
        }
    } else {
        if (colored) {
            image[(x + y * IMAGE_WIDTH) / 8] &= ~(0x80 >> (x % 8));
        } else {
            image[(x + y * IMAGE_WIDTH) / 8] |= 0x80 >> (x % 8);
        }
    }
}

void send_command(unsigned char command) {
	PORTB &= ~(1 << PINB1);
	send_recieve_spi_byte(command);

}

void send_data(unsigned char data) {
	PORTB |= (1 << PINB1);
	send_recieve_spi_byte(data);

}

void epaper_clear(unsigned char* image) {
	for (int x = 0; x < EPD_WIDTH; x++) {
        for (int y = 0; y < EPD_HEIGHT; y++) {
            epaper_draw_absolute_pixel(image,x, y, 0);
        }
    }
}

void epaper_init() {
	DDRD &= ~(1 << DDD1);
	DDRB |= (1 << DDB1);
	send_command(BOOSTER_SOFT_START);
	send_data(0x17);
	send_data(0x17);
	send_data(0x17);
	send_command(POWER_ON);
	while(!(PIND &&(1<<PD1)));
    send_command(PANEL_SETTING);
	send_data(0x8F);
	send_data(0x8F);
    send_command(VCOM_AND_DATA_INTERVAL_SETTING);
    send_data(0x37);
    send_command(RESOLUTION_SETTING);
    send_data(0x68);
    send_data(0x00);
    send_data(0xD4);
}

void epaper_display() {
	send_command(DISPLAY_REFRESH);
	while(!(PIND &&(1<<PD1)));
}

void epaper_set_partial_black(unsigned char* buffer_black, int x, int y, int w, int l) {
	send_command(PARTIAL_IN);
	send_command(PARTIAL_WINDOW);
	send_data( x & 0xf8); // X on 8 kordne
	send_data(((x & 0xf8) + w -1) | 0x07);
	send_data(y >> 8);
	send_data(y & 0xff);
	send_data((y + l -1) >> 8);
	send_data((y + l - 1) & 0xff);
	send_data(0x01);
	_delay_ms(2);
	send_command(DATA_START_TRANSMISSION_1);
	if (buffer_black != NULL) {
        for(int i = 0; i < w  / 8 * l; i++) {
            send_data(buffer_black[i]);  
        }  
    } else {
        for(int i = 0; i < w  / 8 * l; i++) {
            send_data(0x00);  
        }  
    }
    _delay_ms(2);
	send_command(PARTIAL_OUT);  
}

void epaper_set_partial_red(unsigned char* buffer_red, int x, int y, int w, int l) {
	send_command(PARTIAL_IN);
	send_command(PARTIAL_WINDOW);
	send_data( x & 0xf8); // X on 8 kordne
	send_data(((x & 0xf8) + w -1) | 0x07);
	send_data(y >> 8);
	send_data(y & 0xff);
	send_data((y + l -1) >> 8);
	send_data((y + l - 1) & 0xff);
	send_data(0x01);
	_delay_ms(2);
	send_command(DATA_START_TRANSMISSION_2);
	if (buffer_red != NULL) {
        for(int i = 0; i < w  / 8 * l; i++) {
            send_data(buffer_red[i]);  
        }  
    } else {
        for(int i = 0; i < w  / 8 * l; i++) {
            send_data(0x00);  
        }  
    }
    _delay_ms(2);
	send_command(PARTIAL_OUT);  
}

```

and calling it out via

```
unsigned char image[1024];

	DDRD |= (1<< DDD0);
	PORTD &= ~(1 << PIND0);
	PORTD |= (1 << PIND0);
	_delay_ms(2000);

	init_spi_oneway();
	epaper_init();
	epaper_clear(image);
	epaper_set_partial_black(image,0,0,32,32);
	epaper_set_partial_red(image, 0,0,32,32);

	epaper_display();
```

This makes a long 32xsomething stripe? In fact, seems that with values close to 16x16 for the w and h of the partial set, it seems to work fine? Am I doing something wrong in the calculations?

zerulin, 2019-11-05:
It seems that your codes are written for 2.13inch e-Paper (B). However, the 2.13inch e-Paper (B) doesn't support partial refresh.

randohinn, 2019-11-06:
@zerulin Okay, tried implementing the full frame refresh method

```
#include "spi.h"
#include <util/delay.h>
#include <avr/pgmspace.h>

// Display resolution
#define EPDWIDTH      104
#define EPDHEIGHT     212

#define IF_INVERT_COLOR 0

// EPD2IN13B commands
#define PANEL_SETTING                               0x00
#define POWER_SETTING                               0x01
#define POWER_OFF                                   0x02
#define POWER_OFF_SEQUENCE_SETTING                  0x03
#define POWER_ON                                    0x04
#define POWER_ON_MEASURE                            0x05
#define BOOSTER_SOFT_START                          0x06
#define DEEP_SLEEP                                  0x07
#define DATA_START_TRANSMISSION_1                   0x10
#define DATA_STOP                                   0x11
#define DISPLAY_REFRESH                             0x12
#define DATA_START_TRANSMISSION_2                   0x13
#define VCOM_LUT                                    0x20
#define W2W_LUT                                     0x21
#define B2W_LUT                                     0x22
#define W2B_LUT                                     0x23
#define B2B_LUT                                     0x24
#define PLL_CONTROL                                 0x30
#define TEMPERATURE_SENSOR_CALIBRATION              0x40
#define TEMPERATURE_SENSOR_SELECTION                0x41
#define TEMPERATURE_SENSOR_WRITE                    0x42
#define TEMPERATURE_SENSOR_READ                     0x43
#define VCOM_AND_DATA_INTERVAL_SETTING              0x50
#define LOW_POWER_DETECTION                         0x51
#define TCON_SETTING                                0x60
#define RESOLUTION_SETTING                          0x61
#define GET_STATUS                                  0x71
#define AUTO_MEASURE_VCOM                           0x80
#define READ_VCOM_VALUE                             0x81
#define VCM_DC_SETTING                              0x82
#define PARTIAL_WINDOW                              0x90
#define PARTIAL_IN                                  0x91
#define PARTIAL_OUT                                 0x92
#define PROGRAM_MODE                                0xA0
#define ACTIVE_PROGRAM                              0xA1
#define READ_OTP_DATA                               0xA2
#define POWER_SAVING                                0xE3
/*
unsigned char epaper_get_image(int width) {
	return malloc(8*width);
}*/

void epaper_draw_absolute_pixel(unsigned char* image, int x, int y, int colored) {
	if (x < 0 || x >= EPDWIDTH || y < 0 || y >= EPDHEIGHT) {
        return;
    }
    if (IF_INVERT_COLOR) {
        if (colored) {
            image[(x + y * EPDWIDTH) / 8] |= 0x80 >> (x % 8);
        } else {
            image[(x + y * EPDWIDTH) / 8] &= ~(0x80 >> (x % 8));
        }
    } else {
        if (colored) {
            image[(x + y * EPDWIDTH) / 8] &= ~(0x80 >> (x % 8));
        } else {
            image[(x + y * EPDWIDTH) / 8] |= 0x80 >> (x % 8);
        }
    }
}

void send_command(unsigned char command) {
	PORTB &= ~(1 << PINB1);
	send_recieve_spi_byte(command);

}

void send_data(unsigned char data) {
	PORTB |= (1 << PINB1);
	send_recieve_spi_byte(data);

}

void epaper_clear(unsigned char* image) {
	for (int x = 0; x < EPDWIDTH; x++) {
        for (int y = 0; y < EPDHEIGHT; y++) {
            epaper_draw_absolute_pixel(image,x, y, 0);
        }
    }
}

void epaper_init() {
	DDRD &= ~(1 << DDD1);
	DDRB |= (1 << DDB1);
	send_command(BOOSTER_SOFT_START);
	send_data(0x17);
	send_data(0x17);
	send_data(0x17);
	send_command(POWER_ON);
	while(!(PIND &&(1<<PD1)));
    send_command(PANEL_SETTING);
	send_data(0x8F);
	send_data(0x8F);
    send_command(VCOM_AND_DATA_INTERVAL_SETTING);
    send_data(0x37);
    send_command(RESOLUTION_SETTING);
    send_data(0x68);
    send_data(0x00);
    send_data(0xD4);

}

void epaper_display() {
	send_command(DISPLAY_REFRESH);
	while(!(PIND &&(1<<PD1)));
}

void epaper_display_frame(unsigned char* frame_buffer_black, unsigned char* frame_buffer_red) {
    if (frame_buffer_black != NULL) {
        send_command(DATA_START_TRANSMISSION_1);
        _delay_ms(2);
        for (int i = 0; i < 32 * 32 / 8; i++) {
            send_data(pgm_read_byte(&frame_buffer_black[i]));
        }
        _delay_ms(2);
    }
    if (frame_buffer_red != NULL) {
        send_command(DATA_START_TRANSMISSION_2);
        _delay_ms(2);
        for (int i = 0; i < 32 * 32 / 8; i++) {
            send_data(pgm_read_byte(&frame_buffer_red[i]));
        }
        _delay_ms(2);
    }
}
```

and

```	epaper_init();
	epaper_clear(image);
	epaper_display_frame(image, image);
	epaper_display();
```

This seems like it should clear the whole display, no? Nothing happens for me, the display does not refresh :(

randohinn, 2019-11-14:
Any ideas @zerulin? I'm also not exactly sure how big the image array should be? The code here seems to suggest 8*width?

hnwangkg-ezio, 2019-11-16:
Please run the demo we provided directly, do not make any changes and try again.

hnwangkg-ezio, 2019-11-16:
The image array size = (image width / 8)  * image height

randohinn, 2019-11-20:
> The image array size = (image width / 8) * image height

That works! No longer does weird stuff, except if I try to epaper_display_frame, then it is mostly garbage, when setting the two partials separately, all works 😄

closed 2019-11-20 by randohinn