waveshareteam/e-Paper #42: Just bought 7.5inch E-Ink Can't get it to work

LiangShuii, 2019-11-13:
Hi all,

As the title says, I have just got through the mail a 7.5inch E-Ink screen. I am running it with a Raspberry Pi and I have connected the e-Paper HAT.

I have downloaded all the relevant libraries and I have used the example python code from the waveshare wiki website and from this github via git clone.

Unfortunately nothing works, the screen has never changed from when I first opened the box. All I get in the terminal is the following:
`
INFO:root:epd7in5 Demo
INFO:root:init and Clear
DEBUG:root:e-Paper busy

`

Any help would be appreciated as I am very keen to get this project up and running!

Kind regards,

James

hnwangkg-ezio, 2019-11-16:
你的问题是否已经解决了？

hnwangkg-ezio, 2019-11-16:
Determine if the connection is correct and whether the screen used is black and white

markbirss, 2019-11-17:
James

Have you enabled SPI with 
sudo raspi-conifg

or by adding to /boot/config.txt

closed 2019-12-31 by hnwangkg-ezio