waveshareteam/e-Paper #45: Library directories include '&' character in path making it difficult to include via pip

txoof, 2019-11-22:
Python pip cannot handle an '&' character when pulling from a git repository:
`pip install -e 'git+https://github.com/waveshare/e-Paper.git#egg=waveshare-epd&subdirectory=RaspberryPi&JetsonNano/python'`

Results in multiple errors when unpacking. There is no reliable way to escape the '&' character.

Suggest removing the '&' character from the path name and changing it to the word "and."

See Pull request #39 
https://github.com/waveshare/e-Paper/pull/39

bashkirtsevich, 2019-11-22:
In my cases, I just copy necessary files from repo and include in mine (with some modifications).
I think this repo should be splitted to two, only python and any other implementations.
You can do it and push it on pypi.

txoof, 2019-11-22:
Of course you can split out just the parts you want, but those parts become stale and are not updated. Submitting them to pypi forces the submitter to maintain them forever.
 There are at least three abandoned and out of date examples of this on pypi right now. I don't want to contribute to that mess. 
Simply removing the `&` character allows everyone to use pup/pipenv on this repo with no extra steps.

bashkirtsevich, 2019-11-22:
Sad, but this repo looks like a junkyard of python and c source. That's should be splitted.

txoof, 2019-11-24:
@bashkirtsevich 
Is there a better library than this one for interfacing with the WaveShare EPDs? I've been looking, but I haven't found one. I've considered rewriting this one, but I don't think I quite have the appropriate level of python FU to do it properly.

hnwangkg-ezio, 2019-11-27:
We will change RaspberryPi & JetsonNano to _JetsonNano in the future

hnwangkg-ezio, 2019-11-27:
i will change RaspberryPi & JetsonNano to RaspberryPi_JetsonNano

txoof, 2019-11-28:
See my pull request - it's already done :)

closed 2019-12-31 by hnwangkg-ezio