waveshareteam/e-Paper #48: Screen not updating

featherbear, 2019-12-08:
I have a 7.5inch Model B display with the e-Paper HAT on a RPi 3

It was working all the way until August 31st, however since then I have powered off the Pi and have not used the display.

Trying today, I am unable to update the screen, however the `epd7in5bc_test.py` test file runs all of the way.

Is this a 'burn-in' issue? I did not clear the screen before powering it off and storing it away.
However if this was the case, I wouldn't think that it would affect the empty white pixels which also do not update

manijak, 2019-12-10:
Make sure the `Spi` config is still enabled on the raspberry after reboot.

featherbear, 2019-12-10:
@manijak I've tried running `raspi-config` and toggling SPI off and on, though it didn't work.

---

I tried with a different Pi (same model, just a different board), and the screen responded.  So it might be a faulty Pi

closed 2019-12-11 by featherbear