waveshareteam/e-Paper #50: Compatibility with Arduino nano 33 BLE

Knowless, 2019-12-15:
Hi everyone,
Hello, everybody,
I would like to use the Arduino Nano 33 Ble. Unfortunately there is a problem with the library pgmspace.h . It can't be used at the moment. Is there a way to bypass the commands or even create support for the library?

This support could be used for the nano ble sense too, because it is the same chip.

Thank you very much.
Best regards,
Knowless

hnwangkg-ezio, 2019-12-27:
There is no good way. If you just need to test the effect, you can remove this header file. Main.c only clears the screen. Your controller flash is too small, and the programming will be very complicated.

closed 2019-12-27 by waveshare

Knowless, 2020-01-07:
Maybe here is a missunderstanding. I don't speak from normal Arduino Nano.
I speaking of the Arduino Nano 33 BLE. It has a lot more memory as the Atmega328P and it works with it too.
So, flash should'nt be the problem. I tried to get it work without success.
I removed the SPI and used a shiftout command to run it on NRF52832. It worked.
The new Nano 33 BLE used a NRF52840 but I don't understand why it didn't worked. There is no change in the display, but I got no error.

Later on I will check with some oscilloscope and compare the signals.
But it would be nice if someone would try it with more experience.

adriensas, 2020-12-21:
Hello @Knowless !
Have you managed to fine a solution ?
I’m running into the same trouble with an arduino nano 33 ble and I’m starting to loose hope