waveshareteam/e-Paper #51: .net core driver support

manijak, 2019-12-17:
Hi,
I was wondering if there are any plans in making support for .net core based libraries & examples for the different displays, similar to what already exists for C & Python? I would be happy to help out & contribute. There are a few custom made .net libs out there made for one or two displays, but it would be great to have wider support.

I've been trying to get the "Waveshare ePaper 7.5" (B)" to run with a .net core application on RaspberryPi 3b, but currently without success. Tested with both "unosquare/raspberryio" & Microsoft's own IoT library "System.Device.Gpio", but the display does not respond to any commands (nothing happens). The provided c & python demos run just fine. 

The driver library I'm replicating is `EPD_7in5bc.c` for RaspberryPi. Anyone tried something like this with .net core or mono? I would really like to find a way to debug the commands, to test if they are successful or not.

hnwangkg-ezio, 2019-12-27:
We currently have no plans to increase support for .net

closed 2019-12-27 by waveshare

mwenk, 2020-01-12:
> Hi,
> I was wondering if there are any plans in making support for .net core based libraries & examples for the different displays, similar to what already exists for C & Python? I would be happy to [...]

Hi, I'm just starting to implement a .net core lib for this scenario. Do you have a public repo so I can take a look at what you've got already?

manijak, 2020-01-12:
I sure do @mwenk it is on my github: https://github.com/manijak/EPaperHatCore

It is a fork from another repo/user that has succsessfully created a lib and run it on a 2.7” waveshare epaper display. In my repo you will also find the driver for 7.5’’ inch epaper, but for me it is not working.

mwenk, 2020-01-12:
I checked the same repo, did not work for me, too.

I own a 7.5" B&W Display and found this repo written for the 3-color 7.5" B Display :
 https://github.com/mwenk/Waveshare.EPaperDisplay  (it's my fork) - 

I've just started implementing support for 7.5" B&W Device, based on the waveshare python example when I noticed that the example app worked out of the box.  The source-base is more modern than the other one, it uses the core system.gpio and spi libs instead of 3rd party ones.

manijak, 2020-01-12:
Thank you for testing my code, good to know that I’m not insane after all 🤣.

Are you saying that the repo you forked and modified works on your 7in5 b&w display? If so that is great! Will have to try it when I get back from my trip 😄  thanks!

mwenk, 2020-01-12:
> 
> 
> Thank you for testing my code, good to know that I’m not insane after all 🤣.
> 
> Are you saying that the repo you forked and modified works on your 7in5 b&w display? If so that is great! Will have to try it when I get back from my trip 😄 thanks!

No, I forked this Repo:

https://github.com/eXoCooLd/Waveshare.EPaperDisplay

This works with thr 7.5" B&W and with Waveshare 7.5inch e-Paper (B)