waveshareteam/e-Paper #54: Help with image rendering

aceisace, 2020-01-11:
Hi @waveshare team,
I've been using the 7.5" E-Paper (B, old model) display for quite some now. For my [project](https://github.com/aceisace/Inky-Calendar), I have been using pretty old driver files (from 2018 I think). Now I wanted to update the driver files, but there are some things I don't understand about the new driver files. Could you please provide some extra information about that?

The three-coloured E-Papers now have a function named `display(self, blackimage, redimage)`. This was previously not the case, i.e. the input would just be one buffer. Why is there a need for 2 buffers now and how can I render an image (png for example) with the same dimensions as the display? Before, the function `get_frame_buffer (or getbuffer now)` would take an image, map each pixel to either red, black and white and save the data in a single buffer. Then the function `display_frame` would render the image on the E-Paper.

hnwangkg-ezio, 2020-01-13:
In the past, it was processed by pictures. Our current writing method is closer to the essence of screen work. The screen is displayed by overlaying black and white pictures and red and white pictures

aceisace, 2020-01-14:
Thanks for the info, @hnwangkg-ezio .
The new function makes things a bit more complicated. I'm wondering if the conversion of an image to a black/white and red/white image could be done by the `display function` as currently, it's pretty difficult to display an image.

For example, this is the input image:
[image]

Then, this image can be split in 2 bmp files (using numpy for example), one for the black and one for the red/yellow pixels):
black ___________________________________________ red_____________________________________________________

[image][image]

Another question, what kind of input image does the function `display(self, blackimage, redimage)`  need? I tried it with PNG files after converting them to mode `'1'` but the display is still blank after the display update.

aceisace, 2020-01-20:
@hnwangkg-ezio 
It's OK, there is no longer a need to keep this issue open. I've managed to implement my approach of spliiting the colours from an RGB image just before creating a buffer.

closed 2020-01-20 by aceisace

andreasciamanna, 2022-01-27:
@aceisace care to share how?

aceisace, 2022-02-06:
I'm currently using two separate images to get the black and red colours shown correctly on the display. Splicing an RGB image is certainly possible, but getting 100% accuracy is very difficult and the results don't look too good on the display either