waveshareteam/e-Paper #55: Improvement/completion of epd1in54_V2 Code

Knowless, 2020-01-13:
Hi everyone,
I saw that there is a new version of 1.54(_V2)" Display and that the old code is not possible to use.
But the new version of .ino code is not complete or not so good example.
There is no function like SetFrameMemory() and in the example is not shown:
- how to update partial 
- how to print Characters
- Draw lines etc and transfer the type "paint" to "epd"

It would be nice if this could be completed or you couldn't say it is the same display with same functionality.

Best regards
Knowless

MartyMacGyver, 2020-02-01:
There's a good amount of information [on the wiki](https://www.waveshare.com/wiki/1.54inch_e-Paper_Module_(B)) in the "Codes description" tab. I was able to use that to do some basic things with this display.

Knowless, 2020-02-03:
Hi MartyMacGyver,
The wiki will not help in this case. I'm familar of using the "old version" of epaper 1.54" and for simple example the function SetFrameMemory() is not implemented in the version 1.54V2 in epd1in54.cpp.

MartyMacGyver, 2020-02-03:
Ah... I see your point. It seems what's implemented varied widely by device and revision.

waldow, 2020-05-02:
Hi Knowless
I put back all the functionality of (V1) . Please have a look
 [https://github.com/waldow/epd1in54_V2_More](url)

Yardie-, 2020-06-02:
Hi Waldow your link doesn't work.
I did track your code down and even though the paint works the image doesn't display correctly.
This is on an epaper that arrived today.
I can't for the life of me understand why all the code adds WaitUntilIdle at the end of the method call then again in the middle of all the code.
This is not a threaded environment. why not just add it at the start of a method and block the code there.
I have implemented this in my code and it vastly improves the reponsiveness of the code.
The Arduino can get on with other jobs. Then when it needs to send something to the epaper it is prepared to wait.
Please Please fix the include spi.h and the include arduino.h
it is Arduino.h and SPI.h
windows my not care but linux does and everytime I download new code I have to fix it.
Sorry for the rant but it is very frustrating.
Something that worked really nicely in the old code no longer works at all.

waldow, 2020-07-10:
@Yardie-   Yes WaitUntilIdle is a unnecessary evil . The whole busy waiting is used a lot .

waldow, 2020-07-10:
epd1in54 V2 code has been updated addressing missing functionality

closed 2020-12-07 by SSYYL