waveshareteam/e-Paper #60: 1.54 in V2

Flavio-da-Brescia, 2020-02-16:
The library does not work, the example is very poor, and this is holding an order of 250 displays.
I fixed the library, but i still need examples for text and shapes.

Please reply asap

dheera, 2020-02-16:
@Flavio-da-Brescia Can you post your fixed library? I'm also having issues fixing the library for raster images.

dheera, 2020-02-16:
@Flavio-da-Brescia Also random question -- does the V2 display require the reset pin to be connected? I've used the V1 displays without the reset pin.

dheera, 2020-02-16:
Update: I have a V2 display that works with some new code I wrote, but a V2.1 board does not work. Does V2.1 use a different protocol from V2?

Kernald, 2020-04-18:
I also have a rev 2.1 board that doesn't do anything, while the samples in this repo work fine for me with a rev 2 (I can only assume, there's no rev number written on it, but there's a RST pin) board - with the same display. That's both using a Raspberry Pi and an Arduino.

waldow, 2020-05-02:
For Arduino, I put back all the functionality of (V1)  to V2. Please have a look
[https://github.com/waldow/epd1in54_V2_More](url)

Qrome, 2020-05-29:
None of the code examples work with my 1.54 (V2.1) -- very frustrating.

Flavio-da-Brescia, 2020-05-29:
I have a company that designs PCBs shields for arduino. I really dont understand HOW DARE YOU sell a product that DOES NOT WORK. If I did that I would not sleep at night!! Get your s...t together waveshare!

Qrome, 2020-05-29:
Update:  I am very disappointed in the 1.54 (V2.1) board.   I have used another shield (white board) and works perfectly with the GxEPD library for Arduino.   I have not been able to use any of the GxEPD library with the new 2.1 revision board.   I finally got the latest sample from the repository here under epd1in54_V2 to flicker and displays part of an image -- none of the text or fonts display.    But the library and example file here are super bad.    Because of this, I will not be selecting this board for use in my latest project.   I am very disappointed.

SSYYL, 2020-09-17:
Hi everyone, The code has been updated and is just as good as V1

closed 2020-12-07 by SSYYL