waveshareteam/e-Paper #64: Python display() is 9x slower than C version

GaryBoone, 2020-03-24:
I added timers to the demo code in this repository and found that the Python code takes 31 seconds to complete a refresh on the 7.5in black/white screen. Using C, it takes 3.6s to refresh. That's a significant difference that seems unlikely to be due to language issues.

The specific demos I compared were:
```
RaspberryPi&JetsonNano/c/examples/EPD_7in5_test.c
RaspberryPi&JetsonNano/python/examples/epd_7in5_test.py
```

Why is the Python code so slow? Can it be sped up?

dev-ng, 2020-03-24:
Not directly on your subject, but I already posted here my experience with Python and RPI Zero vs RPI 4. With Zero it was very slow about 40-50 seconds to complete refresh. RPI 4 does it within 1-2 seconds. So just wondering what is your hardware with 31s?

GaryBoone, 2020-03-25:
Good call. You are correct. It is an RPI Zero W that yielded those refresh times.

tito, 2020-03-27:
I use is too, but it's not 31/40/50s, the display itself is around 3s. Make sure to differenciate the loading time of -all- Python libraries VS the execution time needed for the display communication/transmission.

LLPP65, 2020-09-14:
I have a long-running Python3 daemon periodically writing to screen. Here are the debug timings on different hardware. While the running time for the display routine is longer on the Pi Zero (15 seconds vs 7 seconds), the flicker/refresh time watching the screens is very similar, about 3.5 seconds.

Raspberry Pi 3 Model B Rev 1.2:
Sep 14 09:46:02: Send image to screen
Sep 14 09:46:02: Start init
Sep 14 09:46:02: End init
Sep 14 09:46:02: Start display
Sep 14 09:46:09: End display
Sep 14 09:46:09: Start sleep
Sep 14 09:46:09: End sleep

Raspberry Pi Zero W Rev 1.1:
Sep 14 09:47:01: Send image to screen
Sep 14 09:47:01: Start init
Sep 14 09:47:02: End init
Sep 14 09:47:02: Start display
Sep 14 09:47:17: End display
Sep 14 09:47:17: Start sleep
Sep 14 09:47:17: End sleep

dev-ng, 2021-01-18:
This is fixed for 7.5in black/white screens with #104 and #130. Sending data byte-by-byte is very slow, especially with Zero.

alex-terry, 2021-02-16:
This change did help with sending the data (cut data send from 1.3 seconds to 0.2 seconds), but for me, the majority of time is spent on the `ReadBusy` command. This means it is still taking ~7 seconds after running `display` function for the final result to show on the screen. Here is my timing using Rasberry Pi 4B, Majaro ARM OS, Python 3.8:

send_data2: 180.38  ms
ReadBusy: 6752.38 ms

closed 2022-11-16 by SSYYL