waveshareteam/e-Paper #66: ModuleNotFoundError: No module named 'spidev'

gavinr, 2020-04-11:
I am trying to run the example `epd_7in5_V2_test.py` for the 7.5" V2 board, getting this error:

```
ubuntu@ubuntu:~/e-Paper/RaspberryPi&JetsonNano/python/examples$ python3 epd_7in5_V2_test.py
Traceback (most recent call last):
  File "epd_7in5_V2_test.py", line 11, in <module>
    from waveshare_epd import epd7in5_V2
  File "/home/ubuntu/e-Paper/RaspberryPi&JetsonNano/python/lib/waveshare_epd/epd7in5_V2.py", line 32, in <module>
    from . import epdconfig
  File "/home/ubuntu/e-Paper/RaspberryPi&JetsonNano/python/lib/waveshare_epd/epdconfig.py", line 146, in <module>
    implementation = RaspberryPi()
  File "/home/ubuntu/e-Paper/RaspberryPi&JetsonNano/python/lib/waveshare_epd/epdconfig.py", line 44, in __init__
    import spidev
ModuleNotFoundError: No module named 'spidev'
```

Environment:
- Raspberry Pi 3 Model B Plus Rev 1.3
- OS: Ubuntu 18 (`ubuntu-18.04.4-preinstalled-server-arm64+raspi3.img`)

bassettb, 2020-04-22:
You need to install it, along with a couple other python packages:
`
pip3 install RPi.GPIO
pip3 install spidev
pip3 install Pillow
`

closed 2020-05-04 by gavinr

objectiveSee, 2020-11-21:
I have the same issue but I tried all the steps that @bassettb suggested. Any suggestions?

```
pi@pi0:~/e-Paper/RaspberryPi&JetsonNano/python/examples $ sudo python epd_2in7_test.py
Traceback (most recent call last):
  File "epd_2in7_test.py", line 11, in <module>
    from waveshare_epd import epd2in7
  File "/home/pi/e-Paper/RaspberryPi&JetsonNano/python/lib/waveshare_epd/epd2in7.py", line 31, in <module>
    from . import epdconfig
  File "/home/pi/e-Paper/RaspberryPi&JetsonNano/python/lib/waveshare_epd/epdconfig.py", line 146, in <module>
    implementation = RaspberryPi()
  File "/home/pi/e-Paper/RaspberryPi&JetsonNano/python/lib/waveshare_epd/epdconfig.py", line 44, in __init__
    import spidev
ImportError: No module named spidev
```

lewisxhe, 2020-11-28:
> I have the same issue but I tried all the steps that @bassettb suggested. Any suggestions?
>
> ```
> pi@pi0:~/e-Paper/RaspberryPi&JetsonNano/python/examples $ sudo python epd_2in7_test.py
> Traceback (most [...]
```
cd ~ 

sudo apt install git -y

git clone https://github.com/doceme/py-spidev

cd py-spidev

sudo python3 setup.py install
```

Tin-Man-Dylan, 2021-03-29:
I had the same issue when I was running "sudo python python epd_4in2bc_test.py" . Then I did "pip install spidev" and it still didn't work.

Later I realized I should be running "sudo pip install spidev", and then it worked running with and without sudo. So seems it was a user privilege thing. So for installing and running these just double-check if you are running it with sudo or not.

waqasnaveed1, 2021-06-09:
sudo pip3 install --upgrade Pillow
in case you are getting No PIL found