waveshareteam/e-Paper #68: 4.2 inch bw e-paper backgroup grayed after Rpi 4 update

gaohoward, 2020-04-16:
Hi, Recently I updated my Ppi 4 system and my 4.2 e-Paper module no longer working properly.
I'm using C api to program the module. 
After update and start my program the screen has a gray (like dust) background where it should be white. (see attached pic)
I didn't change my code so I guess the underlying driver (I'm using wiringpi lib) has changed.
[image]

Yardie-, 2020-04-26:
I have the same problem with mine.
I have the same unit 
 I pulled it out of the box again the other day and I though I had somehow killed it but maybe not.
I ran a clean version of the default code and it was the same.
Updates are written but all grey

Sorry I have no idea what is wrong here but like you I would like to find out.

Yardie-, 2020-06-09:
I just had one idea that might help.
I saw somewhere that you need to store the unit with a completely white screen when not in use.
If it has been a while since you used the epaper then maybe set it to white then leave it then a few days later do a test. It is just an untested idea but it might work.

swkim01, 2020-07-24:
Please update wiringPi to latest version and rebuild.
`cd /tmp`
`wget https://project-downloads.drogon.net/wiringpi-latest.deb`
`sudo dpkg -i wiringpi-latest.deb`

closed 2022-11-16 by SSYYL