waveshareteam/e-Paper #69: The samples don't seem to work with rev2.1 boards

Kernald, 2020-04-18:
I have two displays (a 5.83" and a 2.13", both bw+r), as well as two driver boards: the one that came with the 2.13" display, without a revision number (but with a RST pin, it's not the first revision, as far as I can tell) and a newer rev2.1 driver board.

I tried the samples with the following combinations:

| Controller  | Driver board | Display | Result |
| ------------- | ------------- | ------------- | ------------- |
| Raspberry Pi 3b  | Pre rev2.1  | 2.13"  | Works  |
| Raspberry Pi 3b  | Pre rev2.1  | 5.83"  | Works  |
| Raspberry Pi 3b  | rev2.1  | 2.13"  | **Does not work**  |
| Raspberry Pi 3b  | rev2.1  | 5.83"  | **Does not work**  |
| Node MCU (Arduino-like) | Pre rev2.1  | 2.13"  | Works  |
| Node MCU (Arduino-like) | Pre rev2.1  | 5.83"  | Works  |
| Node MCU (Arduino-like) | rev2.1  | 2.13"  | **Does not work**  |
| Node MCU (Arduino-like) | rev2.1  | 5.83"  | **Does not work**  |
| Arduino Uno | Pre rev2.1  | 2.13"  | Works  |
| Arduino Uno | Pre rev2.1  | 5.83"  | Works  |
| Arduino Uno | rev2.1  | 2.13"  | **Does not work**  |
| Arduino Uno | rev2.1  | 5.83"  | **Does not work**  |

For all the Raspberry Pi combinations, I ran both the Python and the C samples. The output in the terminal is similar regardless of whether it actually does anything or not - it's just faster when it doesn't work. I also tried different cables, the HAT connector... nothing changes. I actually have _two_ rev2.1 boards and have the exact same issue with both, which makes me think that it isn't a hardware issue (or I'm really not lucky).

Did something change between the rev2.1 and the previous one, which makes the samples incompatible? Any advices on what I could try to pinpoint the issue?

ericcorporation, 2020-05-08:
The same happens to me with the display 2.9" rev2.1 on Raspberry Pi Zero WH. The screen goes blank and does not draw anything.
@Kernald Have you tried the test for display 2.13" rev2.1 called **epd_2in13_V2_test.py**? For display 2.9 there is no test V2. I have seen some difference in the function _TurnOnDisplay(self)_ of the library **epd2in13_V2.py**

       self.send_command(0x22)
       self.send_data(0xC7)
       self.send_command(0x20)        
       self.ReadBusy()

In the previous version there was the following:

        self.send_command(0x22) # DISPLAY_UPDATE_CONTROL_2
        self.send_data(0xC4)
        self.send_command(0x20) # MASTER_ACTIVATION
        self.send_command(0xFF) # TERMINATE_FRAME_READ_WRITE
        logging.debug("e-Paper busy")
        self.ReadBusy()
        logging.debug("e-Paper busy release")

technimad, 2020-08-05:
I have a 7.5 in Yellow screen, a rev 2.1 driver board and rpi4, the samples also don't work for me.

SSYYL, 2020-09-17:
Mostly code has update.
if not work please cut down reset() RST low level  delay time.

denisjoshua, 2020-12-14:
Dear Kernald,
I also have a 2.13 black and red e-paper. 
So I use the epd2in13bc library (I think this is the one I must use) 

But I so that in that library is missing the partial refresh part 
https://github.com/waveshare/e-Paper/tree/master/Arduino/epd2in13bc

Can you tell me please what source you use in order to have partial refresh work ? 
Thanks a lot
Denis

SSYYL, 2020-12-14:
@denisjoshua  Hello, I'm sorry to tell you that there is no partial refresh on the three-color screen.

joseph-abell, 2021-02-03:
Hi, I've got a version 2.1 board of 2in13 yellow e-paper. I've tried Python, C, examples, all examples print `E-Paper busy.` I've tried on a pi zero and a pi 4.

Herrner, 2021-02-21:
I'm also unable to use the examples for a 4.2inch 400x300 three-color ePaper module (version 2.1). The `epd_4in2_test.py` example works, but can't make use of the second color. The `epd_4in2b_V2_test.py` (which seems to be the right one) does not work.

closed 2022-11-16 by SSYYL