waveshareteam/e-Paper #72: Using DHT11 arduino sensor with 2in13 HAT e-paper

New-Basic, 2020-05-20:
Hello, i´m a newbie in programming. I need same help to read and send data (temp. and Humidity) from DHT11 to the e-paper display. I can send text to the e-paper and i can read from the sensor in separed project codes but i don't know how i can do it in just one project. 
Thank you

Yardie-, 2020-06-03:
No one seems to answer for ages around here.
If you look at the old examples you will see how to write text to the e-paper the old way. 
A lot of the old stuff is not implemented in the V2 code and it is deffinetly not newbie friendly.
If you have a 2.9 inch e-paper you are in luck. The old code works. so look at the example and send the string to the epaper that you would have sent to Serial.print.
If you haven't got that far yet.
First try and get it printing what you want to the Serial interface.
Then check the examples

New-Basic, 2020-06-04:
> No one seems to answer for ages around here.
> If you look at the old examples you will see how to write text to the e-paper the old way.
> A lot of the old stuff is not implemented in the V2 code and [...]

Hello,
thank you for your reply, i have a 1.54 inch and a 2.13inch e-paper. The data that is aquire from the DHT11 sensor is a float and the code for e-paper print a string. If i convert the float in to a string can it make usable for the e-paper? i'm stuck with this, i have read lots of things but i quit confused by now...
Once again thank you for your help.

Yardie-, 2020-06-06:
Ok 

sprintf() will do the conversion for you
first get your head around that
 here's a start 
```
char out[24];
 int num = 42;
 sprintf( out, "The answer is %d", num );
 Serial.println(out);
```
Once you get the output you want printing to the Serial terminal.

Then download Waldo Wolmarans @waldow version at
https://github.com/waldow

Yes it is also broken a bit also but not so badly and it has the necessary info to get the 1.54 e-paper working the way you want it.

By the time you have done that come back and I will have a better explanation of how to get things working
Greg

New-Basic, 2020-06-07:
Once again thank you for your help Greg. i will try it...i will be back with some feedback. 

Bruno

closed 2022-11-16 by SSYYL