waveshareteam/e-Paper #74: Supporting larger displays than 7.5inch?

Norfeldt, 2020-05-22:
I'm considering getting a 12.48" display, but I don't see this repo supporting larger displays than 7.5". Is there any plans on adding them?

If, then when?

mgpepe, 2020-05-22:
Same issue here. Python unlocks so much more and those screens are crippled if we don't have the libraries.

Norfeldt, 2020-05-31:
@hnwangkg-ezio thank you for recently adding 7.5 inch V3 and HD display support. Any plans for larger displays?

martinberlin, 2020-07-27:
If any of you uses ESP32 I added a class in the component to drive the 12.48" epaper display: 
https://github.com/martinberlin/cale-idf/wiki/Model-wave12i48.h

It's on ESP-IDF framework and I have no plans to make it for Arduino framework (for now)
The problem is that in order to use this epaper, if you want to drive it with an ESP32, you need to use one with PSRAM. Since it has a very large pixel buffer (about 160 Kb)
Sadly the adapter is for a ESP32-devKitC and I didn't found so far one that fits and has PSRAM. Even if it would the design uses the GPIOs that are supposed to be used with internal PSRAM so it won't work as is. 
But the class does work and supports GFX so you can use geometric functions and fonts. A good solution would be make an adapter to use an ESP32 model that supports PSRAM or to use an S2 version.
@waveshare it would be great that on next design, when you place ESP32 pins, this is taken in account. Since with a big GFX buffer, you have little RAM, for the rest of your program.

missionfloyd, 2020-09-10:
https://github.com/waveshare/12.48inch-e-paper

Norfeldt, 2020-09-10:
thank you @missionfloyd 

I however dumped the idea since these displays are way overpriced - it's much cheaper to to buy a portable (color) dislay

closed 2020-09-10 by Norfeldt

martinberlin, 2020-10-09:
For those wanting to add a WROOVER esp32 with PSRAM here is my hack:

https://github.com/martinberlin/cale-idf/wiki/Model-wave12i48.h#esp32-wroover-hack-for-this-epaper

By the way thanks @waveshare for linking my component from the display Wiki page. I think this epaper work great for any office environment and they have a nice size and technology. I really like how it works with ESP32 / ESP-IDF since you don't need to attach a highly more consuming RaspeberryPI to it.