waveshareteam/e-Paper #75: Please include a license

b6d, 2020-05-22:
Hi,

I would like to use your code in an open source project but I'm unable to do so because your repository doesn't specify a license and I can't find one anywhere else.

Could you please specify & include a license for your project?

Thank you!

(more information: https://help.github.com/en/github/creating-cloning-and-archiving-repositories/licensing-a-repository)

TrashcanCoder, 2020-06-09:
Hello,

i want to use it in an commercial product for one customer would be nice to have a license, too.

Thx

caemor, 2020-08-24:
> I would like to use your code in an open source project but I'm unable to do so because your repository doesn't specify a license and I can't find one anywhere else.

Look into the files you are using there is a license included in nearly every code file, but most likely this full repo will never get a single license since there is also code from arm and stm included.