waveshareteam/e-Paper #77: V2 is upside down.

Yardie-, 2020-06-03:
Does anyone know what is with the orientation of the V2 e-papers are they upside down or what is going on.
Thanks.

waldow, 2020-06-07:
On the _Driver output control_ setting there are a _direction_  flag.
I modified it back like in V1
````
    SendCommand(0x01); //Driver output control
    SendData(0xC7);  
    SendData(0x00);  
    SendData(0x0); //change direction was  0x01
````
It makes updating the full screen code easier but break the partial (SetMemoryArea/SetMemoryPointer)

Yardie-, 2020-06-08:
haha yes that works I did find that in bit in your code .
Your fixes are very helpful.
What you did ie via.
`  byte1 =  image_buffer[i + (j-y) * (image_width / 8)];`
is also required or the direction is still weird.
Also the white/black concept is opposite to that expected in V1

With the 2.9 inch V1  I have a clock which

- a few seconds before the minute ->full update init 
- displays buffer 1
- clears buffer 0 to white
- writes buffer 0 black text on white (main information)
- on the minute -> partial update init
- displays buffer 0
- clears buffer 1 to white
- writes to buffer 1 white on black (alternative information)

This then does a full refresh each minute and sets the pixels to their opposite (no ghosting). 
The alternate information then justifies the required refresh flash.
This two buffer approach does not seem possible in version 2.

It seems that it has been sacrificed for apparent speed however I can't see the speed improvement as you have to write to both buffers each time. Which seems very wasteful. Especially using the default approach that does waits after ever call instead of before.

Thank you very much waldow you code was invaluable.

waldow, 2020-06-08:
Yes, the 2 buffer was first only for the multi color displays like the red and black displays. One for black and then the other one for red.

I experienced the same thing when the 2.13 screen went to V2.

They maybe went to the 2 buffers register to do gray scale that can be cool. 

There is some weird toggling happening that maybe cause that makes it invert.

To minimize the ghosting I do a short delay and do another update with the same data.

I see in the datasheet that is a bit confusing there are modes , maybe it can be used to set it back to work as before on one buffer.
Try 0XB9 was 0XB1
```
    SendCommand(0x22); // //Load Temperature and waveform setting.
    SendData(0XB9);
    SendCommand(0x20);
```

Yardie-, 2020-06-09:
Ok I'll give that a try thanks again @waldow .
Which datasheet is that. It may be easier to decipher than the code.
I am primarily a c coder and some of the cpp stuff just seems a lot like like going around in circles to me. And lots of stuff that isn't good in a single threaded environment.

Yardie-, 2020-06-11:
Thanks again @waldow after a little rtfm I am starting to understand this a little better.
I have been thinking that the epapers where like the early crt drivers with at least two display buffers. If the red buffer is available on all units perhaps my approach will work once I understand it all a little better and work out how to use it like that. 
I have been going about this the wrong way. better to understand the interface than the waveshare code. thanks.

waldow, 2020-07-10:
epd1in54 V2 code has been updated addressing missing functionality and only use one buffer when necessary.