waveshareteam/e-Paper #80: stuck on DEBUG:root:e-Paper busy

deguilardi, 2020-06-23:
By simply running `python3 epd_7in5_V2_test.py`, I get stuck on DEBUG:root:e-Paper busy.
Any command triggers DEBUG:root:e-Paper busy.
by simply calling init methods, I am stuck.
    epd.init() or epd.Clear() won't work.

raspberry zero, e-paper 7.5in V2

Choromanski, 2020-07-11:
I am having the same issue with my 2.13 in v2.1

iia, 2020-07-20:
@deguilardi I have the exact same issue. My hardware is identical to yours as well.

mastern2k3, 2020-07-24:
> By simply running `python3 epd_7in5_V2_test.py`, I get stuck on DEBUG:root:e-Paper busy.
> Any command triggers DEBUG:root:e-Paper busy.
> by simply calling init methods, I am stuck.
> epd.init() or epd.Clear() won't work.
> 
> raspberry zero, e-paper 7.5in V2

exactly the same issue here, and with the exact model and script - the 7.5in V2

iia, 2020-07-26:
As this problem was bugging me the whole weekend I tried the same hardware (Display + HAT) with a Raspberry Pi 3B+. And all the problems were gone! The "stuck in busy state" was gone and it worked like a charm! I was also having trouble with the HAT as in running the test script was causing the Pi Zero W to reset (no idea why).

Now why is this happening with a Pi Zero W and not with a 3B+? To me it seems like because the Pi Zero W has a armv6 and 3B+ a armv7 CPU. Some of the involved software libraries or maybe even the HAT works better with (was designed for) armv7.

If this is the case Waveshare must update their documentation and really mention this issue instead of assuming that everything works in Pi Zero with the HAT.

sfPlayer1, 2020-08-02:
dmesg reports:
> [ 2322.191616] spidev spi0.0: setup: unsupported mode bits 8
which is SPI_LSB_FIRST

Oddly enough every other implementation than dev_hardware_SPI.c I've checked uses MSB first, same with the datasheet and user manual.

Removing https://github.com/waveshare/e-Paper/blob/master/RaspberryPi%26JetsonNano/c/lib/Config/dev_hardware_SPI.c#L95 switches this implementation to MSB first as well, which appears to work fine with the RPi Zero W.

iia, 2020-08-02:
@sfPlayer1 Good catch! But this will fix the issue only with the C examples and not with the Python examples, right?

I think for the Python examples a Python module called `spidev` is used which handles the SPI operations using IOCTL through `/dev/spidev0.0 `. Same usage (and version) of this `spidev` python module works in Pi 3 B+ but not in Pi Zero W for me.

SSYYL, 2020-12-03:
Hello, everyone. 
Has this problem been solved? I guess it may be a bug caused by our hardware update. 
At present, the software has been fully updated. If there are any questions, please send an new issue and I will answer them in time

kretep, 2020-12-27:
Did anyone find a solution to this problem? I also have a 7.5 inch V2. The C version of the demo works, but the Python version mostly results in DEBUG:root:e-Paper busy. The strange thing is that of the ~40 times that I tried, the Python demo worked exactly once... without me doing anything notably different (at least that I was aware of).

kretep, 2020-12-29:
Interestingly, it does work with Python when connecting the board to the gpio header using the wires instead of directly putting it on top. This is less than ideal considering the extra space it takes, so I would still like to put it on top. What could cause the different behavior?

sfPlayer1, 2020-12-29:
Interesting observation. The most likely explanation involves different I/O pin use, should be easy enough to measure/verify.

kretep, 2020-12-29:
@sfPlayer1 I'm not sure how to measure that; can you give some pointers?

I have some more observations. With HAT on top:
* First time after power on: resets at `epd.init()`
* Second time: works and keeps working until `epdconfig.module_exit()` is called
* After that it is stuck in `epd.init()` with `DEBUG:root:e-Paper busy`, even after reboot
* Disconnect power and start from top

With the wires it just always works.

tesslinger, 2020-12-29:
@kretep was the HAT working at the beginning? I observed the issue that it was working properly, but two weeks later I have the problem as described here.

kretep, 2020-12-29:
@tesslinger I've had this problem since I started using it two or three days ago...

SSYYL, 2020-12-30:
Hello everyone,

The problem of restarting at Pi ZERO is caused by our  e-Paper Driver HAT. The hardware has been updated.
Look at this #92

sfPlayer1, 2020-12-30:
@kretep my guess appears to be incorrect, i assumed the python code may have been using different pin assignments, which you rectified by wiring it up differently.

It now seems that the wires just act as extra impedance decoupling the board from the Pi, basically acting like a small inductor and resistor. This works around the apparent stability issue in the Pi's power supply. Changing the capacitors as linked by SSYYL sounds plausible. Considering the 5V come from the external USB power supply, you may have some room for improvement there as well, i.e it may be inadequate.

SSYYL, 2020-12-30:
This seems to be more than just a RPI ZERO restart problem.
I just verified on RPI ZERO that the code did have this problem.

kretep, 2020-12-31:
That is unfortunate. I could just live with the double boot time and not call `epdconfig.module_exit()`, but I wonder if that is not damaging the HAT and/or rpi. Is it? I will also try some other power supplies.

Superberti, 2021-01-08:
I discovered same kind of problems with the 2.7" HAT, too. After power on my pi 0 often reboots in epd.init(). And after calling epd.Dev_exit() it gets stuck in epd.init(). Is it possible that it suffers from the same problem?

pdavid-muc, 2021-01-18:
I have the exact same issue. Cannot run the samples on my pie 0 with the 7.5 inch V2 display.

pdavid-muc, 2021-01-18:
> 
> 
> Hello everyone,
> 
> The problem of restarting at Pi ZERO is caused by our e-Paper Driver HAT. The hardware has been updated.
> Look at this #92
> 
> modification:
> This seems to be more than just a RPI ZERO restart problem.
> 
> I just verified on RPI ZERO that the code did have this problem.
> Python code sometimes does not work properly.

I've bought my HAT off Amazon - do i need to replace the HAT module?

SSYYL, 2021-01-19:
@pdavid-muc , Please try to use the PH2.0 8PIN interface.
The current version only addresses the Pi Zero reboot issue, and other non-working issues are currently being tested.

pdavid-muc, 2021-01-19:
@SSYYL thank you for the hint - is there a guide on the wiring? I don't know whit wire has to go to which pin.

SSYYL, 2021-01-20:
@pdavid-muc Here are some useful information: [waveshare wiki](https://www.waveshare.com/wiki/7.5inch_e-Paper_HAT)

Peerke1969, 2021-02-07:
Hello everyone,

I had the same issue for some weeks, I have bought a 7.5 inch V2 and started testing according Waveshare in Python and got stuck on "DEBUG:root:e-Paper busy".
Bought then a 2.13 inch e-paper HAT for testing which worked directly :-).

Start testing with the 7.5 inch and searched the software for the busy signal without result.
Just started testing all 7.5 inch versions and guess what it was a HD three color version.
So it worked with the following test:  **sudo python epd_7in5b_HD_test.py**

Good luck all and stay healthy

TheTechRobo, 2021-05-30:
I have the Same issue on Raspberry Pi 3B. Haven't tested on zero or 4 since id ont have either.

treii28, 2021-07-25:
I am getting this with the red+black 2.13 inch on a pi zero W. One thing I note is that I can run the c-code demo just fine, but python3 version and/or my own attempts at writing a short demo always return 'busy'.

hubmartin, 2021-09-21:
Hi, I did some investigation on my RPI Zero W since I had `e-Paper busy` issues sometimes too.
This might not apply to all issues people mention above, but I've discovered two things during my project:

- The voltage over 40 pin connector is different than if you use separate wires (5 V on connecgor vs 3.3V on wires) - BUSY signal gives 5 V signal to the RPI which might destroy pin or reboot PI. PI has 3.3 V pin tolerant pins, not 5 V.
- Python has 2 ms delay in reset function, which might take longer on underpowered Zero, which disconnects power for epaper and code is stuck on `e-Paper busy`.

More detailed analysis on my blog https://www.martinhubacek.cz/blog/waveshare-epaper-hat-issues/

kretep, 2021-12-05:
@hubmartin good catch! I finally managed to get rid of these pesky wires by redirecting the 3.3V!

lombre55, 2022-12-28:
Hi, I have the same problem. After few messages with waveshare support, they send a new hat.
I also bought a new one, but I think the new is burned too.
Or, it's my 7.5 inch B/W screen (waveshare too). It's all grey.

nekopoolgithub, 2023-04-01:
> @kretep was the HAT working at the beginning? I observed the issue that it was working properly, but two weeks later I have the problem as described here.

Exactly the same issues years later. I'm using a Raspberry Pi 3
INFO:root:epd7in3f Demo
INFO:root:init and Clear
DEBUG:waveshare_epd.epd7in3f:e-Paper busy H
[image]

I could play with the ePaper screen for 2 weeks then it shut down for good.

One thing to mention is that the refresh was REALLY BRUTAL. Like flashing for like 5-10 seconds straight before showing an image. Could this be 
a faulty unit ?

My project WAS looking fine !
[image]

SaschaHeyer, 2023-04-04:
reporing the same issue with 7.5inch e-paper V2 and Raspberry Zero.
Any updates here?

It's a shame that we never get a response from waveshare. Could be such a nice product but you fail massivly with the support and documentation.

rikosintie, 2023-04-17:
"By simply running python3 epd_7in5_V2_test.py, I get stuck on DEBUG:root:e-Paper busy."

I had the same issue with a 2in13 V3 hat mounted in the GPIO pins. I was running
python3 epd_2in13b_V3_test.py

I tried 
python3 epd_2in13_V3_test.py

And it came right up. Notice that the second string doesn't have the "b" after 2in13.

nekopoolgithub, 2023-04-17:
On m’y case the script was running for like 2 weeks and suddenly stopped Sent from Proton Mail for iOS

Dinu2226, 2023-05-16:
Did somebody found a solution for this? I have the same problem.

Dinu2226, 2023-05-16:
> The unit was replaced. There is no solution for this, this is an hardware problem.
> […](#)
> ___________________________ Thomas ZYNDA ***@***.*** +33 6 50 47 82 28 Sent with Proton Mail secure email.
> ------- Original Message -------

Which unit you mean? What hardware component should I replace to make the project work?

TheTechRobo, 2023-05-16:
This can also occur if you did not connect the display correctly to your device.

Dinu2226, 2023-05-18:
> This can also occur if you did not connect the display correctly to your device.

I have an raspberry pi b+ v1.2, the connectors look fine, I dont really know what should I do, I need the setup for the banchelor stathesis, I cant get another components în time due to the microcontroler crysis. Any piece of adviece?

SaschaHeyer, 2023-05-18:
I had two devices side by side and agree one was defect.

kestasjk, 2023-06-18:
I was also getting the paper busy message with the 1.54 inch unit. I was running the wrong demo from following the install guide I google'd. 

sudo epd_1in54b_test.py works, python epd_1in54b_V2_test.py doesn't.

It's unfortunate but it seems the paper busy message can be for a lot of reasons.

sepastian, 2023-08-01:
This seems to work:

1. use the cables to connect the display; don't use the pHAT, it will power the display with 5V instead of 3.3V. Thanks for finding out and for the [detailed analysis](https://www.martinhubacek.cz/blog/waveshare-epaper-hat-issues/) @hubmartin! 
2. Make sure to run the correct example. I own a 2.7in b/w display, so I'm running `python examples/epd_2in7_test.py`. If in doubt, simply try all the examples for your screen size (4 for the 2.7in display, i.e. `epd_2in7_*py`).

bloomingeek, 2024-02-05:
For what's it's worth, here was my solution for my Waveshare 7.5 B&W e-paper:
I was using a Lexar 16GB micro "B" HC card, which is about eight years old. With this card I never achieved a test screen. Always the dreaded "busy" response. (I'm using a Raspberry Pi Zero 2 W with the HAT that came with the screen.)

Now I'm using a SanDisk 32GB "10" A1 card, newly purchased. Screen now works using the epd_7in5_V2_test_old.py and the epd_7in5_V2_test.py.

I'm guessing the older SD card was too slow? (It handled the Raspberry Pi OS and setup with no trouble.) Hope this helps!

schmijos, 2024-05-19:
> One thing to mention is that the refresh was REALLY BRUTAL. Like flashing for like 5-10 seconds straight before showing an image.

This why I landed here. It eventually shows the image but flashes like hell pouring debug messages:

```
Clear...
Debug: e-Paper busy H
Debug: e-Paper busy H release
Debug: e-Paper busy H
Debug: e-Paper busy H release
Debug: e-Paper busy H
Debug: e-Paper busy H release
Goto Sleep...
```

Yifferpi, 2024-09-02:
I had the exact same problem, 2in13 V4 display on a RPi 3B+ connected via wires. My fix was switching the VCC from a 5V pin to a 3.3V pin. One resource I had used showed a pinout which used a 5V pin but the waveshare wiki suggested 3.3V.

Hope it helps someone :)