waveshareteam/e-Paper #81: waveshare 7.5inches v2 won't draw

deguilardi, 2020-06-23:
I'm just trying to run any test to see if the screen draws anything.
By simply running `pi@raspberrypi:~/Workspace/e-Paper/RaspberryPi&JetsonNano/python/examples $ python3 epd_7in5_test.py` I get all logs as it's all good, but there is absolutely nothing drawn on the screen.

The screen is all white.

logs:
INFO:root:epd7in5 Demo
INFO:root:init and Clear
DEBUG:root:e-Paper busy
DEBUG:root:e-Paper busy release
DEBUG:root:e-Paper busy
DEBUG:root:e-Paper busy release
INFO:root:1.Drawing on the Horizontal image...
DEBUG:root:1234
DEBUG:root:imwidth = 640  imheight =  384 
DEBUG:root:e-Paper busy
DEBUG:root:e-Paper busy release
INFO:root:2.Drawing on the Vertical image...
DEBUG:root:1234
DEBUG:root:imwidth = 384  imheight =  640 
DEBUG:root:e-Paper busy
DEBUG:root:e-Paper busy release
INFO:root:3.read bmp file
DEBUG:root:1234
DEBUG:root:imwidth = 640  imheight =  384 
DEBUG:root:e-Paper busy
DEBUG:root:e-Paper busy release
INFO:root:4.read bmp file on window
DEBUG:root:1234
DEBUG:root:imwidth = 384  imheight =  640 
DEBUG:root:e-Paper busy
DEBUG:root:e-Paper busy release
INFO:root:Clear...
DEBUG:root:e-Paper busy
DEBUG:root:e-Paper busy release
DEBUG:root:e-Paper busy
DEBUG:root:e-Paper busy release
INFO:root:Goto Sleep...
DEBUG:root:e-Paper busy
DEBUG:root:e-Paper busy release
DEBUG:root:spi end
DEBUG:root:close 5V, Module enters 0 power consumption ...

dev-ng, 2020-06-23:
Hi, I own 7.5" v1. It has native resolution 640x384. You've got v2. As far as I know it comes with 800x480. There is also HD version with 880x528.
Looking at the log you provided you are using script compatible with v1. It's actually clear from your post: epd_7in5_test.py

Try epd_7in5_V2_test.py

It has:
# Display resolution
EPD_WIDTH       = 800
EPD_HEIGHT      = 480

sh969, 2020-06-27:
Had the same issue. Worked by selecting 7in5_V2 for my 800x480 display (two colour).

RobinMW, 2020-07-07:
I have the same issue here. I’ve connected it both manually with the jumper cables and with just the hat attached to my Raspberry Pi 3B+ and tested all versions of epd_7in5... but none of them worked. The code runs but nothing changes on the display. Could there be an other reason for that? I have the Waveshare 7.5” black-white-red display 640-384.

sh969, 2020-07-07:
You have the colour version but neither V2 nor HD so it should be epd7in5bc. The files in the lib folder define the resolution.

RobinMW, 2020-07-08:
Ah thanks for that! But should I be able to run the epd7in5bc.py in the lib folder? Because when I run it, it gives ImportError: cannot import name ‘epdconfig’ from __main__’ (epd7in5bc.py). And if I run the epd7in5bc_test.py in /examples, the code runs but nothing happens on my display. What might I be doing wrong here?

sh969, 2020-07-08:
Interesting. When you say the code runs do you get the full console output? When I run my example (via _sudo python epd_7in5_V2_test.py_ for my version) I get:

**INFO:root:epd7in5_V2 Demo
INFO:root:init and Clear
DEBUG:root:e-Paper busy
DEBUG:root:e-Paper busy
INFO:root:1.Drawing on the Horizontal image...
DEBUG:root:Vertical
DEBUG:root:e-Paper busy
INFO:root:2.Drawing on the Vertical image...
DEBUG:root:Horizontal
DEBUG:root:e-Paper busy
INFO:root:3.read bmp file
DEBUG:root:Vertical
DEBUG:root:e-Paper busy
INFO:root:4.read bmp file on window
DEBUG:root:Horizontal
DEBUG:root:e-Paper busy
INFO:root:Clear...
DEBUG:root:e-Paper busy
DEBUG:root:e-Paper busy
INFO:root:Goto Sleep...
DEBUG:root:e-Paper busy
DEBUG:root:spi end
DEBUG:root:close 5V, Module enters 0 power consumption ...**

If yes, maybe the error is not software related. Since you are using the HAT there is little chance the GPIO connections are connected wrongly (maybe a loose soldering patch) but you could check if all the molex display connections on the HAT cables are inserted correctly.

RobinMW, 2020-07-08:
Thanks for your reply. I've checked the connections both via the HAT and with just the cables and nothing changes. 
Here is my console output when I run epd_7in5bc.py from /examples:

**INFO:root:epd7in5bc Demo
INFO:root:init and Clear
DEBUG:root:e-Paper busy
DEBUG:root:e-Paper busy release
DEBUG:root:e-Paper busy
DEBUG:root:e-Paper busy release
INFO:root:Drawing
INFO:root:1.Drawing on the Horizontal image...
DEBUG:root:imwidth = 640  imheight =  384 
DEBUG:root:Horizontal
DEBUG:root:imwidth = 640  imheight =  384 
DEBUG:root:Horizontal
DEBUG:root:e-Paper busy
DEBUG:root:e-Paper busy release
DEBUG:root:e-Paper busy
DEBUG:root:e-Paper busy release
INFO:root:2.Drawing on the Vertical image...
DEBUG:root:imwidth = 384  imheight =  640 
DEBUG:root:Vertical
DEBUG:root:imwidth = 384  imheight =  640 
DEBUG:root:Vertical
DEBUG:root:e-Paper busy
DEBUG:root:e-Paper busy release
DEBUG:root:e-Paper busy
DEBUG:root:e-Paper busy release
INFO:root:3.read bmp file
DEBUG:root:imwidth = 640  imheight =  384 
DEBUG:root:Horizontal
DEBUG:root:imwidth = 640  imheight =  384 
DEBUG:root:Horizontal
DEBUG:root:e-Paper busy
DEBUG:root:e-Paper busy release
DEBUG:root:e-Paper busy
DEBUG:root:e-Paper busy release
INFO:root:4.read bmp file on window
DEBUG:root:imwidth = 640  imheight =  384 
DEBUG:root:Horizontal
DEBUG:root:imwidth = 640  imheight =  384 
DEBUG:root:Horizontal
DEBUG:root:e-Paper busy
DEBUG:root:e-Paper busy release
DEBUG:root:e-Paper busy
DEBUG:root:e-Paper busy release
INFO:root:Clear...
DEBUG:root:e-Paper busy
DEBUG:root:e-Paper busy release
DEBUG:root:e-Paper busy
DEBUG:root:e-Paper busy release
INFO:root:Goto Sleep...
DEBUG:root:e-Paper busy
DEBUG:root:e-Paper busy release
DEBUG:root:spi end
DEBUG:root:close 5V, Module enters 0 power consumption ...**

Nothing seems to be going wrong here I think. Might I have a faulty HAT or display? Or are there maybe other libraries or demos that will work with my display, just to test if it actually works?

technimad, 2020-08-05:
Hi, 
I also have the color version, not V2, and I am experiencing the same issues. Nothing is displayed, yet the test runs ok.
The screen isn connected vo the HAT. Dipswitches are in the following positions: 
Display config: B 
Interface config: 0

This is the output I get from the test script:

```
INFO:root:epd7in5bc Demo
INFO:root:init and Clear
DEBUG:root:e-Paper busy
DEBUG:root:e-Paper busy release
DEBUG:root:e-Paper busy
DEBUG:root:e-Paper busy release
INFO:root:Drawing
INFO:root:1.Drawing on the Horizontal image...
DEBUG:root:imwidth = 640  imheight =  384 
DEBUG:root:Horizontal
DEBUG:root:imwidth = 640  imheight =  384 
DEBUG:root:Horizontal
DEBUG:root:e-Paper busy
DEBUG:root:e-Paper busy release
DEBUG:root:e-Paper busy
DEBUG:root:e-Paper busy release
INFO:root:2.Drawing on the Vertical image...
DEBUG:root:imwidth = 384  imheight =  640 
DEBUG:root:Vertical
DEBUG:root:imwidth = 384  imheight =  640 
DEBUG:root:Vertical
DEBUG:root:e-Paper busy
DEBUG:root:e-Paper busy release
DEBUG:root:e-Paper busy
DEBUG:root:e-Paper busy release
INFO:root:3.read bmp file
DEBUG:root:imwidth = 640  imheight =  384 
DEBUG:root:Horizontal
DEBUG:root:imwidth = 640  imheight =  384 
DEBUG:root:Horizontal
DEBUG:root:e-Paper busy
DEBUG:root:e-Paper busy release
DEBUG:root:e-Paper busy
DEBUG:root:e-Paper busy release
INFO:root:4.read bmp file on window
DEBUG:root:imwidth = 640  imheight =  384 
DEBUG:root:Horizontal
DEBUG:root:imwidth = 640  imheight =  384 
DEBUG:root:Horizontal
DEBUG:root:e-Paper busy
DEBUG:root:e-Paper busy release
DEBUG:root:e-Paper busy
DEBUG:root:e-Paper busy release
INFO:root:Clear...
DEBUG:root:e-Paper busy
DEBUG:root:e-Paper busy release
DEBUG:root:e-Paper busy
DEBUG:root:e-Paper busy release
INFO:root:Goto Sleep...
DEBUG:root:e-Paper busy
DEBUG:root:e-Paper busy release
DEBUG:root:spi end
DEBUG:root:close 5V, Module enters 0 power consumption ...
```

technimad, 2020-08-05:
just confirmed the c example also doesn't output anything to the screen.

```
./epd 
EPD_7IN5BC_test Demo
/***********************************/ 
Current environment: Raspbian
set wiringPi lib success !!! 
/***********************************/ 
e-Paper Init and Clear...
Debug: e-Paper busy
Debug: e-Paper busy release
Debug: e-Paper busy
Debug: e-Paper busy release
NewImage:BlackImage and RYImage
show window BMP-----------------
pixel = 100 * 100
Debug: e-Paper busy
Debug: e-Paper busy release
Debug: e-Paper busy
Debug: e-Paper busy release
show red bmp------------------------
pixel = 640 * 384
pixel = 640 * 384
Debug: e-Paper busy
Debug: e-Paper busy release
Debug: e-Paper busy
Debug: e-Paper busy release
show image for array
Debug: e-Paper busy
Debug: e-Paper busy release
Debug: e-Paper busy
Debug: e-Paper busy release
EPD_Display
Debug: e-Paper busy
Debug: e-Paper busy release
Debug: e-Paper busy
Debug: e-Paper busy release
Clear...
Debug: e-Paper busy
Debug: e-Paper busy release
Debug: e-Paper busy
Debug: e-Paper busy release
Goto Sleep...
Debug: e-Paper busy
Debug: e-Paper busy release
free(): invalid next size (normal)
Aborted
```

tanahy, 2020-08-08:
I also have problems to draw anything at all on the 7in5_V2, but in my case the example file is more explicit about the errror:

`sudo python3 epd_7in5_V2_test.py `

```
INFO:root:epd7in5_V2 Demo
INFO:root:init and Clear
DEBUG:root:e-Paper busy
DEBUG:root:e-Paper busy
INFO:root:cannot open resource
```

I get two flashes of black and white and then the program ends due to the error.

markolson, 2020-08-08:
I'm having the same issue with the [7.5 color + hat](https://www.waveshare.com/7.5inch-e-Paper-HAT-C.htm) and the included examples. Going to email email support and see if they have anything helpful to say 🤷

tanahy, 2020-08-08:
So I solved the issue I had by installing the libraries using the setup.py file, which is missing in the docs. Before doing so I had to manually set the library on the same folder as the examples but it didn't work properly as I explained before.

I would advise the dev to include the following instructions in the readme_rpi_XX.txt files, the same files for the jetson and in the wiki:
```
cd python 
sudo python3 setup.py build
sudo python3 setup.py install
```

markolson, 2020-08-10:
Unfortunately that didn't work for me, but I do feel a bit dumb that I didn't use `setup.py` earlier.

mchen0037, 2020-08-24:
I'm getting this error too, was anyone able to fix?

jfreax, 2020-08-24:
Same display, some problems. I think that the v2 version just doesn't work with the current examples.

mchen0037, 2020-08-25:
Is there any indicator or file I could use to verify that this is a code issue instead of a hardware issue?

tanahy, 2020-08-26:
@mchen0037 I would recommend running the C program because it's just easier to follow the steps but in any case I was able to draw correctly with the python demo as well by using the setup.py commands I commented on before.

I also updated a small library for micropython in case you have an esp32 or similar and want to try it (check the dev branch): [https://github.com/tanahy/micropython-waveshare-epaper/tree/dev](https://github.com/tanahy/micropython-waveshare-epaper/tree/dev)

jfreax, 2020-08-26:
@tanahy Thanks for trying to help!
Neither the C, nor the Python Version works for me. My display just keeps being blank. Tried with a Raspberry Pi and an Arduino.

But I think, I probably have a completely different version. It's clearly marked as "V2" and it's 7.5" in size. That's all I know for sure. I bought it as (WaveShare 7.5inch E-Ink Display HAT for Raspberry Pi 880×528 Red / Black / White (WS17960)" from https://eckstein-shop.de/WaveShare-75inch-E-Ink-Display-HAT-for-Raspberry-Pi-880528-Red-Black-White

As you can see, it should have a resolution of 880×528 pixel, red / black / white and is version 2. None of the example matches this specification. Also none of the datasheets provided by waveshare...

markolson, 2020-08-26:
@jfreax that’s the exact model I ordered (direct from waveshare), and they said I really ordered the “HD” model and should use that code. It still didn’t work for me, but could be worth trying.

jfreax, 2020-08-26:
@markolson I tried all 7in5* example codes. No luck yet. I also wrote an email to waveshare's support team. Let's see if they can help.

mchen0037, 2020-08-26:
Thanks @tanahy

I'm in the same boat as @jfreax. Neither C nor Python work out, but I also wrote them an email.

pixelspark, 2020-09-02:
Same issue here! Interested to hear support's reply...

markolson, 2020-09-02:
Last communication I got back from them said to try using wires instead header pins as a connection. so I'm pretty close to giving up. :-\

jfreax, 2020-09-03:
I got an answer from the support. They told me, that my device is in fact a "HD" version, but that there is no "V2" variant of the HD devices. Otherwise, they just send me a link to their wiki. Not that helpful.

My screen is clearly labeled as "V2" on the backside. So, I'm not sure who is right here. But regardless of which version it is, I tested all 7in5 examples. On different devices, C and Python... It just doesn't work.

@markolson I'm also pretty close to giving up :(

pixelspark, 2020-09-03:
@jfreax interesting, mine has a sticker saying "SKU: 14144. 7.5inch e-Paper (C)". What I ordered is the kit with a HAT (SKU 14229), listed as `640x384, 7.5inch E-Ink display HAT for Raspberry Pi, yellow/black/white three-color`. The resolution seems to imply this to be a V1 panel.

technimad, 2020-09-03:
> Last communication I got back from them said to try using wires instead header pins as a connection. so I'm pretty close to giving up. :-\

Hi,

Did you try connecting using wire pins?
I did, and now the examples work!

Details of my setup:
rpi4b
SKU: 14144. 7.5inch e-Paper (C)
HAT rev2.1

Connected not directly as Hat to the gpio headers, I use the supplied wires, connected to the hat:
[image]

Reference for the pinout of the wires:
- https://pinout.xyz/pinout/spi
- https://www.waveshare.com/wiki/7.5inch_e-Paper_HAT_(B) (Hard- software tab)

Don't know why this works, and not when directly connected. But it seems to work as expected.

pixelspark, 2020-09-03:
I did actually use the wires, with an ancient Raspberry Pi original model B. The old Pi might be the culprit (will try with a Pi Zero as soon as I can find the header to solder to the board).

markolson, 2020-09-03:
I used wires on my Pi Zero W, but will try with a Pi B+ this weekend

mchen0037, 2020-09-09:
I just tried wires again, no luck. I'm using a Rasp Pi 4 Model B.

waveshare, 2020-09-09:
Hello everyone, I want to say sorry first because we do not check the GitHub all the time and it is hard for us to reply to issues. If you have any question want to confirm with us directly, please send an email to service@waveshare.com.

waveshare, 2020-09-09:
As you know, we have three kinds of 7.5inch e-Paper. The oldest one is the V1 version with 640x384 resolution and it stops producing.
The demo codes for the V1 version should be epd_7in5_test.py and epd_7inch5bc_test.py.
The V2 version has 800x480 resolution and the codes should be epd_7in5_V2_test.py and epd_7in5b_V2_test.py
The HD version has 880 x 528 resolution and the codes should be ped_7inch5_HD_test.py and epd_7in5b_HD_test.py.

Please confirm the type of e-Paper you bought and use the correct codes.

waveshare, 2020-09-09:
And please check if you have connected the e-Paper panel in the correct way, we found that some of the users reverse the cables and it causes the problem.
Please set the Display Config switch to B sides.
[image]

If you check that all the cables and confirm that all are connected properly, please clone the newest demo codes and try again. And make sure that the SPI interface was enabled and doesn't use for other drivers or devices.

waveshare, 2020-09-09:
> Hi,
>
> Did you try connecting using wire pins?
> I did, and now the examples work!
>
> Details of my setup:
> rpi4b
> SKU: 14144. 7.5inch e-Paper (C)
> HAT rev2.1
>
> Connected not directly as Hat to the gpio [...]

Do you have a multimeter to test if there are shorted pins in the black connector of the driver board?

pixelspark, 2020-09-16:
I have connected my Pi Zero directly to the HAT (using the header). The display is a V1, 640x384, 7.5 inch one (SKU 14229). I followed the instructions from the wiki linked above exactly, on a fresh install of Raspbian Buster. Still does not work. 

[image]

[image]
[image]
[image]

Note that I have already tried connecting using wires to *another* Raspberry Pi (a Model B in that case) - also no dice.

I am suspecting a production issue with the screen... @waveshare how do I arrange an RMA?

technimad, 2020-09-17:
Yes, I have a multimeter. What do you want me to test? Which combination of pins should I check?

fred2nice, 2020-11-30:
I have a 7.5 3 colors 880x528, Raspberry and e-paper Driver Hat.
With Python examples no way, same behaviour.
With C the same.
But If I change from WIRING to BCM inside Makefile
USELIB_RPI = USE_BCM2835_LIB
#USELIB_RPI = USE_WIRINGPI_LIB

It WORKS with EPD_7in5b_HD_test(); !
( have to run twice the first time ) 
I test with Raspberry Pi3 and Zero W, same.

pixelspark, 2020-11-30:
> I have a 7.5 3 colors 880x528, Raspberry and e-paper Driver Hat.
> With Python examples no way, same behaviour.
> With C the same.
> But If I change from WIRING to BCM inside Makefile
> USELIB_RPI = [...]

I quickly checked it here - it does seem to make some difference (i.e. the code for EPD_7in5b_HD does not hang like it did before) but still nothing on the screen. This would also be pretty strange given my display should be a V1, not an HD. Note of the other 7in5-test functions appear to work either (they do hang). Will try using a Zero W later.

fred2nice, 2020-11-30:
> I quickly checked it here - it does seem to make some difference (i.e. the code for EPD_7in5b_HD does not hang like it did before) but still nothing on the screen. This would also be pretty strange [...]

I am still discovering how this device works ...
My raspberry is configured with these parameters :
dtparam=i2c_arm=on
#dtparam=i2s=on
dtparam=spi=on
The device worked only after I changed to USE_BCM2835_LIB then make clean;make; reboot, then run epd.
If I test python scripts, then run again the C version nothing on screen, It works again after un run / ctrl+c, run again epd binary or a reboot ....

But still no way with Python ( I tested several other projects with E-Paper and python, still withe screen ).

If anyone has a simple python script for 7 "5 3 colors (880x528) a simple" Hello world  displayed "will light up my night ...

Jianhong-Lu, 2020-12-02:
My device is 4b,unlike @pixelspark  ,several days ago,I could do like this article.
https://debugger.medium.com/how-to-build-a-very-slow-movie-player-in-2020-c5745052e4e4
With test.mp4 or the new video I tranfer to raspi,it can go very well.So I turn off the power,and the picture is remained(that is ok,the e-paper traits)
[image]
[image]
  
But when I got up,running the helloworld or slowmovie,it failed ,the picture remains the same,when I tried to connect up or down. 2 errors are below.
 
[image]
[image]

So I debug the epd_7in5_V2_test.py ,found after it log “1.xxx “ ,the program can’t go well,in 31 row (Maybe this row, tomorrow I will check again),maybe this casue the “debug root..”?
 
[image]

Also the e-paper adapter has one pin called “BUSY”,so I check the 7.5 specifications:
 
[image]

Maybe the previous picture dominate the e-paper some state ,so it can’t start code?

@technimad And also I use voltmeter to test 2 pins, if I run the .py program ,it is 0,but if it stops ,I can get 2V approximately.
 
[image]

pixelspark, 2021-01-09:
So, progress on my side! I recently ordered a 7in5b V2 display (b/w, no colors) with HAT. Much to my surprise, this worked out of the box! 

Turns out that the 7in5bc (v1) display now works as well using the newly received HAT (and the appropriate demo code). Still a bit glitchy (and the colors are less pronounced than I would expect) but at least something I can work with.

@waveshare looks like the first HAT was DOA, would you be able to send me a replacement?

homeallone, 2021-02-03:
> @tanahy Thanks for trying to help!
> Neither the C, nor the Python Version works for me. My display just keeps being blank. Tried with a Raspberry Pi and an Arduino.
>
> But I think, I probably have a [...]

Hi, where you ever able to resolve this? I bought the same recently ... did not do the math before I bought it. Anyway did you get the display working?

nodepond, 2021-03-19:
I still have issues on this, like described in thread Thread with a 7in5b HD with 880x528.

I tried before with a custom made SPI-interface in an Arduino-style environment when I percieved this kind of issue before. Debugging send me down this road. Testing with Raspberry Pi 400, connected via Wires to e-Paper HAT. Triple checked everything. Also tried different displays / HATS, cables etc. It just hangs at at epd_7in5b_HD_test.py at 

INFO:root:init and Clear
DEBUG:root:e-Paper busy

Than nothing. The V2 examples run through, but no display.

It would be cool to hear any news on this. We are using them for an art project with deadline approaching and let's say... We spent some money on the hardware, so it would be cool to make them run. The e-Inks I tried before were an out of the box experience, and this here is very frustating, since I already spent 3 days of debuggin...

nodepond, 2021-03-19:
Is there maybe something like a firmware-update we could try?

tanahy, 2021-03-20:
Did you try my fork of the library? It's made for micropython but it 
should work with little to no modification on the raspberry PI, take a 
look at my repo in case it's of any help, it worked properly on my V2 
screen.

nodepond, 2021-03-22:
Hey, thankt @tanahy for your answer. Didn't make it work so far, because I am not familiar with MicroPython and looks like a lot of fiddle to get something to work quickly.

Anyway. I found something new, that I find some sort of confusing.

On the webpage of the 7.5inch_e-Paper_HAT_(B) you find the following notice: (https://www.waveshare.com/wiki/7.5inch_e-Paper_HAT_(B))

[image]

And then a just link to a shop page. 

1. What do they mean with "Software Update"?
2. In the wiki I found a dedicated page to a 7.5inch_HD_e-Paper_HAT_(B) (notice the HD in the name!!!!). Link: https://www.waveshare.com/wiki/7.5inch_HD_e-Paper_HAT_(B)

But again no specific infos about a HD version...

I ordered my HAT a the bundle withthe eInk at a supplier in Germany, maybe there are newer versions of the HAT available, but you can't tell the difference between them from the outside?

And why can't they link the page of the HD version in the notice above I screenshotted...

Can anyone bring some more light into this issue? Is there a firmware-update available for the HAT or not? If not, are there 2 different versions of the HAT available (no HD and HD)? If yes, how can we tell the difference between both versions?

nodepond, 2021-03-22:
I am in exactly the same situation a @jfreax in this thread. Ordered the same hardware at the same supplier. Could you meanwhile sort something out @jfreax?

nodepond, 2021-03-29:
Today I made a setup, that worked. So it is not an issue related to the hardware itself.

The current setup was:
- Rasberry Pi 3B+
- e-Paper Drive HAT Rev 2.1
- Python example from waveshare (7in5b_test_HD)

The most significant difference to the previous setups is, that I used this time the HAT directly on the Raspberry-Pi pins and not the SPI-Cable. So it seems, that the issues are somehow related to SPI. That would be more or less to opposite of this comment from Waveshare, suggesting using SPI instead of Header-Pins (https://github.com/waveshare/e-Paper/issues/81#issuecomment-686541468).

[image]

[image]

CrankyDad, 2022-01-09:
I had the same symptoms as described. In my case the c program works as intended, but the Python example did not. The script just executed, no warnings but nothing was drawn. I use a Raspberry 3B on Bullseye
I saw different behavior looking at the signals on the GPIO pins that there were no noticeable traffic on the SPI pins when running the Python script but a lot of traffic when running the C program.

I did a test to reduce the frequency for  the SPI communication (in epdconfig.py) by
...
    RST_PIN         = 17
    DC_PIN          = 25
    CS_PIN          = 8
    BUSY_PIN        = 24

    **SPI_FRQ = 2000000**
...
 #       self.SPI.max_speed_hz = 4000000
        **self.SPI.max_speed_hz = self.SPI_FRQ**
        logger.debug("SPI frequency:"+str(self.SPI_FRQ))

Now the Python script works as intended and draws on the screen.
The Hat (rev 2.2) is still wonky and stop working after a while. Connecting via cables does the trick...

jimmy-malarkey, 2022-02-28:
I am not able to get my 7.5 inch 3 color display to do anything.  Im using a Raspberry Pi Zero and the display that I bought from Amazon.  It is supposed to be a White / Black / Red at 800×480.  I have tried running all of the 7in5 examples, but nothing is drawn on the screen.  I am trying to figure out for sure which version I have, but mine has a V3 sticker on the back.  I cant find any documentation that such a thing even exists?
[image]
I  tried it with the HAT and I  tried with the wires.  I  reinstalled Raspberry Pi Lite multiple times.  I stepped through the Python instructions on the wiki over and over.  Most of the test scripts act like they are running (Drawing on the Horizontal image...) but nothing is on the screen.  

Can anyone help me out or point me in the right direction?  This is very frustrating.

CrankyDad, 2022-03-01:
Had the same issue on my B/W model. In my case the transfer rate of the SPI was too high.Test the Python example provided by WaveShare for your display to ensure that it works with Python programsThere might be an issue where the Python program runs but nothing is draw to the display. If so, try modify  in epdconfig.py (in lib folder) to the following SPI_FRQ = 2000000 (this reduces the data rate over SPI)Does the C-program work? The issue above relates to the Pyhon code only.There is also software that shows the activity on the pins too see if there is any communication. The HAT is not reliable and I only use wires to the relevant pins. /Jan

jimmy-malarkey, 2022-03-12:
I have a working display now.  The problem for me was a damaged Pi board.  I opened a ticket with Waveshare and they confirmed that V3 does exist and its functionally the same as V2.  The V2 demo code works on V3.  I wired mine up using the 8 pins and the 'Board' column of the documentation.  

For me the process to get the Python code to run was:

1. Fresh install of Raspberry Pi OS Lite
2. Enable SPI
3. Install Python3 libraries
          sudo apt-get update
          sudo apt-get install python3-pip
          sudo apt-get install python3-pil
          sudo apt-get install python3-numpy
          sudo pip3 install RPi.GPIO
          sudo pip3 install spidev
4.  Install Git
          sudo apt-get install git
5.  Download demo code
          sudo git clone https://github.com/waveshare/e-Paper
6. Run the demo
          sudo python3 epd_7in5b_V2_test.py

Hope this helps someone.

Martin-H-42, 2022-04-10:
My issue was that after running the C example first, the python example did not display anything without doing a reboot. Otherwise fine, just follow the instructions from the wiki and add the setup.py build steps from @tanahy (see above).

My setup: RaspberryPi 2b+, Latest Raspberry Pi OS 2022-04-04, 7.5inch V2 display with HAT.

Bazz-Man, 2022-09-13:
> I have a working display now. The problem for me was a damaged Pi board. I opened a ticket with Waveshare and they confirmed that V3 does exist and its functionally the same as V2. The V2 demo code [...]

So I have the same unit (numbers on the back exactly the same and with V3 sticker) and also had the same issue, no working display.  But followed @jimmy-malarkey instructions above and worked fine.  This is as of 12th Sept 2022 and was using bullseye 32bit release.  Also I noted that python3, python3-pil, python3-numpy, RPi.GPIO and git were already installed and at the latest version.  Also I didnt need to run setup.py either.  
My switches are set to display config 'B' and interface coding to '0' on the interface board and also connected it via jumper cables to the PI zero W.  Also worked fine on a PI Zero 2 W and also plugging it as a HAT on the PI zero.

Hope that helps anyone still struggling with this great display (when it works of course!) 👍

saifis, 2023-01-10:
I had a V2 and a pi zero and the C demo worked fine but the python code did not following the instructions, turns out all I needed to do was reboot the pi and it worked just fine.

anuragphadke, 2023-02-06:
I have a pi zero 2w and I followed the pin-out diagram as per https://www.youtube.com/watch?v=QU4LxcYIbuE (unsure if its same for pi zero and pi-zero 2w)
- confirmed SPI interface is enabled.
- followed the tutorial steps (with clean image etc.)
- on running 
sudo ./epd
EPD_7IN5_V2_test Demo
/***********************************/
Current environment: Raspbian
bcm2835 init success !!!
/***********************************/
e-Paper Init and Clear...
Debug: e-Paper busy
---------
Display continues to stay blank.

Any one has a pin-out for connecting to pi zero 2w? Wasn't able to find anything from the official website (already emailed support, no reply)

dwehrmann, 2023-03-14:
> 

I've been following the instructions, installation of the python modules and everything went clean without any errors. 
However, when I try to run the example code, I get

`Traceback (most recent call last):
  File "/home/pi/e-Paper/RaspberryPi_JetsonNano/python/examples/epd_7in5_V2_test.py", line 11, in <module>
    from waveshare_epd import epd7in5_V2
  File "/usr/local/lib/python3.9/dist-packages/waveshare_epd-0.0.0-py3.9.egg/waveshare_epd/epd7in5_V2.py", line 32, in <module>
    from . import epdconfig
  File "/usr/local/lib/python3.9/dist-packages/waveshare_epd-0.0.0-py3.9.egg/waveshare_epd/epdconfig.py", line 222, in <module>
    implementation = JetsonNano()
  File "/usr/local/lib/python3.9/dist-packages/waveshare_epd-0.0.0-py3.9.egg/waveshare_epd/epdconfig.py", line 113, in __init__
    raise RuntimeError('Cannot find sysfs_software_spi.so')
RuntimeError: Cannot find sysfs_software_spi.so`

Screen stays empty, of course.

hueyy, 2023-04-09:
@dwehrmann Did you copy over the `sysfs_software_spi.so` file from the `lib/waveshare_epd` folder?

owardlaw, 2023-10-27:
After about 5 hours of debugging, I solved the issue of no display in Python. I needed an older OS. 

I downloaded [2020-05-28 version](https://downloads.raspberrypi.com/raspios_armhf/images/raspios_armhf-2020-05-28/version) of the OS and went through the [docs](https://www.waveshare.com/wiki/7.5inch_e-Paper_HAT_Manual) from waveshare. It then worked instantly in python. 

--------------------
Ras pi 3
7.5 inch V2 dispay

kanefron5, 2026-02-11:
The same issue: `INFO:root:cannot_ open resource`. Any workaround?
This worked with C, but not with Python.

Jay-Madden, 2026-03-08:
I was having a whole heck of a lot of issues getting the python epd_7in5_V2_test.py file to work on an older 32 bit rpi zero w

The C driver however did work, ended up diffing the two and enabled `SPI.no_cs = True` in the python version and that fixed it. Fill diff here 

```diff
--- a/RaspberryPi_JetsonNano/python/lib/waveshare_epd/epdconfig.py
+++ b/RaspberryPi_JetsonNano/python/lib/waveshare_epd/epdconfig.py
@@ -55,7 +55,7 @@ class RaspberryPi:
         self.SPI = spidev.SpiDev()
         self.GPIO_RST_PIN    = gpiozero.LED(self.RST_PIN)
         self.GPIO_DC_PIN     = gpiozero.LED(self.DC_PIN)
-        # self.GPIO_CS_PIN     = gpiozero.LED(self.CS_PIN)
+        self.GPIO_CS_PIN     = gpiozero.LED(self.CS_PIN)
         self.GPIO_PWR_PIN    = gpiozero.LED(self.PWR_PIN)
         self.GPIO_BUSY_PIN   = gpiozero.Button(self.BUSY_PIN, pull_up = False)

@@ -72,11 +72,11 @@ class RaspberryPi:
                 self.GPIO_DC_PIN.on()
             else:
                 self.GPIO_DC_PIN.off()
-        # elif pin == self.CS_PIN:
-        #     if value:
-        #         self.GPIO_CS_PIN.on()
-        #     else:
-        #         self.GPIO_CS_PIN.off()
+        elif pin == self.CS_PIN:
+            if value:
+                self.GPIO_CS_PIN.on()
+            else:
+                self.GPIO_CS_PIN.off()
         elif pin == self.PWR_PIN:
             if value:
                 self.GPIO_PWR_PIN.on()
@@ -90,8 +90,8 @@ class RaspberryPi:
             return self.RST_PIN.value
         elif pin == self.DC_PIN:
             return self.DC_PIN.value
-        # elif pin == self.CS_PIN:
-        #     return self.CS_PIN.value
+        elif pin == self.CS_PIN:
+            return self.CS_PIN.value
         elif pin == self.PWR_PIN:
             return self.PWR_PIN.value

@@ -143,6 +143,7 @@ class RaspberryPi:
             self.SPI.open(0, 0)
             self.SPI.max_speed_hz = 4000000
             self.SPI.mode = 0b00
+            self.SPI.no_cs = True
         return 0

     def module_exit(self, cleanup=False):
@@ -157,7 +158,7 @@ class RaspberryPi:
         if cleanup:
             self.GPIO_RST_PIN.close()
             self.GPIO_DC_PIN.close()
-            # self.GPIO_CS_PIN.close()
+            self.GPIO_CS_PIN.close()
             self.GPIO_PWR_PIN.close()
             self.GPIO_BUSY_PIN.close()
```

Im not entirely sure why this happens but this did seem to fix it for me. Hope it helps someone else.