waveshareteam/e-Paper #82: 2.7英寸墨水屏硬件上支持局刷,但python库的驱动还没有支持

lixy123, 2020-06-24:
1. 根据以下资料，2.7英寸墨水屏硬件上支持局刷,但微雪官网却显示不支持局刷
https://item.taobao.com/item.htm?spm=a1z1r.7974869.0.0.269c3ad4K3lf8j&id=558318378111
我在esp32下用GxEPD库调试，发现局刷效果很好。希望你们在python上增加局刷函数.
另外，看了e-Paper的实现代码，GxEPD的局刷是真的局刷，只传输需要刷的数据。而e-Paper的局刷是传全部的数据，然后用局刷模式。显示屏越大，二种算法的速度区别会越大。
2.4.2英寸墨水屏硬件也支持局刷，看你们python库代码的注释提到其效果不好，放弃了局刷功能。 效果确实差一点，但在1米开外是可以接受的，建议把局刷功能加上，让用的人决定是否用局刷功能.

hnwangkg-ezio, 2020-12-11:
代码应该是支持了，只不过是当时没有更新到github。

lixy123, 2020-12-11:
> 代码应该是支持了，只不过是当时没有更新到github。
看了您的新Python库源码
1.4.2英寸墨水屏有局刷函数了,虽然硬件局刷能力略差,毕竟多了一种程序选择. 
2.2.7英寸墨水屏仍然没看到有新加的局刷支持函数,在arduino环境 2年前就已有支持局刷的库了  https://github.com/ZinggJM/GxEPD 有空还是加上吧.
3.局刷传入的图像数据是全屏的图像图像,效率不高.

michael2020-new, 2021-03-09:
> 1. 根据以下资料，2.7英寸墨水屏硬件上支持局刷,但微雪官网却显示不支持局刷<br/>
> 这是厂家介绍<br/>
> https://item.taobao.com/item.htm?spm=a1z1r.7974869.0.0.269c3ad4K3lf8j&id=558318378111<br/>
> 这是微雪官网介绍<br/>
> http://www.waveshare.net/wiki/2.7inch_e-Paper_HAT<br/>
> 
> 我在esp32下用GxEPD库调试，发现局刷效果很好。希望你们在python上增加局刷函数.
> 另外，看了e-Paper的实现代码，GxEPD的局刷是真的局刷，只传输需要刷的数据。而e-Paper的局刷是传全部的数据，然后用局刷模式。显示屏越大，二种算法的速度区别会越大。<br/>
> 2.4.2英寸墨水屏硬件也支持局刷，看你们python库代码的注释提到其效果不好，放弃了局刷功能。 效果确实差一点，但在1米开外是可以接受的，建议把局刷功能加上，让用的人决定是否用局刷功能.

您好！目前我也在Arduino和ESP32上尝试实现2.7 inch墨水屏的局部刷新功能，但在微雪没有找到相关局刷例程，请问您能帮忙提供一下吗？感谢！

lixy123, 2021-03-11:
@michael2020-new  好久没用这个了, 如果用ESP32,  你好可试试GxEPD, GxEPD2 这二个开发库，有2.7 inch墨水屏的局部刷新功能的demo.

michael2020-new, 2021-03-11:
> @michael2020-new  好久没用这个了, 如果用ESP32,  你好可试试GxEPD, GxEPD2 这二个开发库，有2.7 inch墨水屏的局部刷新功能的demo. 
> 

好的 谢谢