waveshareteam/e-Paper #92: Waveshare 7.5" crashes and reboots raspberry pi zero

coredumper2020, 2020-09-05:
Hello,

Using the [7.5inch E-Ink display HAT for Raspberry Pi, red / black / white three-color](https://www.waveshare.com/product/7.5inch-hd-e-paper-hat-b.htm) crashes and reboots the [Raspberry Pi Zero W](https://www.raspberrypi.org/products/raspberry-pi-zero-w/).

It crashes precisely there:

```
(Pdb) bt
  ./main.py(11)<module>()
-> epd.init()
  ./e-Paper/RaspberryPi&JetsonNano/python/lib/waveshare_epd/epd7in5b_HD.py(80)init()
-> self.reset()
> ./e-Paper/RaspberryPi&JetsonNano/python/lib/waveshare_epd/epd7in5b_HD.py(50)reset()
-> epdconfig.digital_write(self.reset_pin, 1)
```

I am able to reproduce without Python nor the e-Paper library.

```shell
echo "17"  > /sys/class/gpio/export 
echo "out" > /sys/class/gpio/gpio17/direction
echo "1"   > /sys/class/gpio/gpio17/value  # system reboots
```

The problem does not appear if the screen is disconnected, or if the screen is plugged to a Raspberry Pi 4.

missionfloyd, 2020-09-09:
Same problem here. Works fine on Pi 4.

Works _sometimes_ on Pi 0.

coredump2020, 2020-09-12:
You are right, I was being imprecise. It is a combination of both the Pi 0 and the HAT.

When you say it stops working, what do you mean by that? Do you also experience the reboot, and everything is fine afterwards? Have you tried using the colored wires?

I have ordered another screen on Amazon to see if I have a faulty unit. I am also discussing with Waveshare support team.

missionfloyd, 2020-09-13:
Yeah, It's rebooting and might work afterwards, but eventually reboots again. Basically exactly as you described.

I tried it with the wires just now, and it seemed to work.

coredump2020, 2020-09-13:
Thanks.

I discussed with someone who has the same setup (Pi0 + Waveshare 7.5 + HAT), but they don't encounter any issue.

I ordered another screen on Amazon and will get it today. I'll share my results!

coredump2020, 2020-09-13:
Same issue with another screen. I updated the description in the ticket and shared my results with their customer service.

missionfloyd, 2020-09-13:
I ordered a new screen as well. I'll let you know if it makes a difference.

xtagon, 2020-09-13:
Same thing happening to me with a screen I just ordered. (Also using 7.5 black/white/red + Pi Zero W + HAT). Looking forward to hearing your results from Waveshare customer service 👍

xtagon, 2020-09-13:
Wiring it up using the wire harness instead of the HAT fixes it for me too. Thank you for opening this issue, it was helpful! 💯

coredump2020, 2020-09-14:
@xtagon you're welcome!

I updated the description with an answer from Waveshare.

missionfloyd, 2020-09-14:
Got out the multimeter. Pins 8, 12, 22, 31, and 36 all appear to shorted to 3V3.

coredump2020, 2020-09-14:
What do you conclude from this? AFAIK only pins 12 and 22 are used by the screen, as GPIOs (so 3.3V is normal).

missionfloyd, 2020-09-14:
I was counting the pins wrong. Those are ground.

coredump2020, 2020-09-14:
Oh so you mean 3.3V is directly sent to the GND? What happens after reboot when the screen works?

missionfloyd, 2020-09-14:
No, I mean the 'shorts' I detected were just the ground pins connected to each other.

missionfloyd, 2020-09-15:
I did notice that the HAT's actually using 5V, not 3.3. So I tried connecting the VCC wire to 5V and it reboots as it did before. It must be drawing more current than the Pi 0 can handle.

coredump2020, 2020-09-17:
Final update (see description). I will leave the issue open, but nothing more can be done.

LLPP65, 2020-09-18:
I'm having the same issue with a black and white 7.5" V2 display. It pretty consistently reboots once per day. I remember seeing a wiring guide to use the SPI interface instead of the hat but I can't remember where. Can someone please point me to an image or wiring guide. Thanks in advance!

xtagon, 2020-09-18:
@LLPP65 Wiki is here: https://www.waveshare.com/wiki/7.5inch_e-Paper_HAT I'm not sure if that's your exact model so you may need to double check. In that page go to Hardward/Software Setup at the top and there is a wiring guide for Raspberry Pi. You can use https://pinout.xyz/pinout/spi for an SPI-specific guide to the Raspberry Pi pins.

LLPP65, 2020-09-19:
@xtagon Thanks for the tip. That doc didn't have all the connections in it but I found the rest in "RaspberryPi&JetsonNano/python/readme_rpi_EN.txt". The screen is currently running OK, fingers-crossed it won't reboot on it's own tonight.

If anyone is interested, I got the following reply from Waveshare support:

```
Hi,

Sorry about the problem you faced. That may be some driver problem when this e-paper working with Pi zero.
You can try to modify the driver board as below:
1.Add 0.1uF capacitor between VCC INT and GND

6TY9YB3PTL5)E3]]CODMM}L.png

2.Remove a 1uF capacitor between VCC and GND, and only reserve 0.1uF capacitor

B0OEM[3KI32L{APKD4HZ84R.png
And the Schematic diagram you can refer to our wiki:https://www.waveshare.com/wiki/File:E-Paper-Driver-HAT-Schematic.pdf 
But if you don't know how to modify the driver board, please contact us again, we can send you a new modifying driver board. 

```
The image files above are at https://padilla.net/oc/index.php/s/WCWrcL47n6s4Lxj

closed 2020-09-20 by coredump2020

LLPP65, 2020-09-20:
Confirmed - using the PH2.0 colored wires and not connecting the hat to the hat pins the Pi-0 did not reboot on its own after en entire day. Now I'm waiting for the modified replacement hat I was promised.

LLPP65, 2020-10-05:
I've been running with the new, fixed hat for three days and it hasn't spontaneously rebooted. It looks like it is fixed, so specify the new hat if you are running a Pi Zero (W).

dbrant, 2020-10-10:
I've actually just received one of these HATs (the crashing version), and found this issue because my Pi Zero was crashing. Thanks for all the suggestions here.
Instead of messing around with capacitors, wouldn't it be simpler to do this:
- remove (clip off / unsolder) the 5V pin(s) from the Pi header
- attach the HAT to the Pi
- solder a wire from the 3.3v pin to VCC on the HAT

missionfloyd, 2020-10-16:
I came up with something similar. An adapter, of sorts. I got a stacking header, cut off the 5V pins, and bent the 3.3V pin so it goes to 5V on the HAT.

deslee, 2020-12-01:
Can someone detail how to connect the colored wires to the pins on the Pi Zero?

xtagon, 2020-12-01:
@deslee Look up your model on the Waveshare wiki, for example the 7.5 inch HD one is here: https://www.waveshare.com/wiki/7.5inch_HD_e-Paper_HAT

Click on "User Guides of Pi" at the top and there will be a pinout table that links the e-paper hat pins to the Pi pins. The HAT pins are labeled on the HAT itself by the connector. For the Pi, you can look at this to know what the names are for each pin: https://pinout.xyz/

fileinster, 2021-05-22:
I have this issue on a 2.13in screen, marked in the back as "Rev2.1". What revision includes the fix for this problem?

emilyboda, 2021-11-19:
Was this issue closed because it was fixed? Because I think I'm still having this issue.

dbrant, 2021-11-19:
@emilyboda This was a hardware issue that impacted a specific version of the e-ink HAT, and the manufacturer acknowledged that there's no software workaround. The previous comments in this issue discuss hardware workarounds that might work for you.

fileinster, 2021-11-21:
I have v2 hardware but still experiencing the same issue.

emilyboda, 2021-11-22:
> I have v2 hardware but still experiencing the same issue.

Do you mean you have v2 of the screen? Or is there a v2 of the HAT? Or.... v2 of the Pi Zero? Because I was thinking of buying the Zero v2, but I won't if this is still a problem.

fileinster, 2021-11-22:
I mean the hat. It has a little holographic "V2" sticker on it, yet my pi
still crashes when trying to set the display if it has been on for a while.
However, for my implementation, the pi resets, and when it reboots it runs
and works correctly. So with a little code manipulation, it makes little
difference apart from a 1-2 minute delay in doing what it needs to do as it
only updates twice daily when no-one is looking. The takeaway from this is
that if you're constantly updating the display, there will be a point where
the Pi will reboot when trying to set the display. How big a deal that is
depends on your use case.

Phil

emilyboda, 2021-12-09:
For the record, I have four 7.5" waveshare displays running on Raspberry Pi Zeros. The displays were bought between 2017 and 2020 and only one HAT has this issue. I swapped out the HAT with the issue for a HAT that was known to be good and everything appears to be fine. To be clear, it was only the HAT I swapped out, I'm using the same Pi, same program, same screen, even the same cable between the HAT and the screen. Is waveshare willing to replace the affected HATs?

xtagon, 2021-12-09:
> Is waveshare willing to replace the affected HATs?

When I asked them regarding my affected HAT order, they said "The new version is not mass-produced yet, sorry about it."

fileinster, 2021-12-09:
I would imagine the answer to this question will be "... ... ...".

At least, that was my mileage when I raised issues with V2 boards.

Phil

emilyboda, 2021-12-09:
Just for more information for people who will be finding this thread later:

- If your HAT has no version number on it, you should have no issues. [See a picture of it here](https://imgur.com/pGEqYNP).
- If your HAT says Rev2.1 on it, you will have issues running it on a Pi Zero. [See a picture of it here](https://imgur.com/eQMb1VV).

I have four HATs working with Pi Zeros, three have no issues and have no version number. The one that says Rev 2.1 is the only one with the issue. Is this consistent with what other people have seen?

All my Pi Zeros are v1; haven't tried it with the v2 Pi Zero that just came out a month ago.

emilyboda, 2021-12-11:
I contacted the company. They've confirmed that they have a new HAT. It is Rev2.2 and it does not have the Pi Zero issues.

emilyboda, 2022-01-02:
Hello all- I've received the Rev 2.2 HAT and confirmed that there are no issues. See pictures [here](https://imgur.com/a/xqfkSN9).