waveshareteam/e-Paper #93: Selection of correctly working partial sizes..

randohinn, 2020-09-07:
I have a bit of a problem deciding on what size to use for my partials, so that the `(x + y * EPDWIDTH) / 8` array location finding formula would not give me a float (flooring and rounding misplaces the pixel or causes weird anomalies). For example, trying to set pixel 20,20 of a 40x40 partial.. Or of a 44x44 partial :/ What's the correct method for making sure in a partial, all pixels are settable?

objectiveSee, 2020-11-25:
@randohinn did you ever solve this?

randohinn, 2020-11-25:
> @randohinn did you ever solve this? 

Not fully no, I guess you either need some rouning for the coordinates or I'm missing something when it comes to alignment...

randohinn, 2020-11-29:
Maybe @waveshare @hnwangkg-ezio  know anything

randohinn, 2021-12-31:
> @randohinn did you ever solve this?

I think I found what I was stupidly ignoring on first implementation of this driver. See https://github.com/waveshare/e-Paper/blob/7003be16721b5f328e21ada1dec18727085c0144/Arduino/epd2in9bc/epdpaint.cpp#L34