waveshareteam/e-Paper #98: Can't write text over a rectangle

hakarune, 2020-09-16:
I'm trying to write text over a rectangle, but only the rectangle displays and no text...

here's the code I'm trying, but it doesn't work...

```
HBlackimage = Image.new('1', (epd.height, epd.width), 255)  # 298*126
HRYimage = Image.new('1', (epd.height, epd.width), 255)  # 298*126  ryimage: red or yellow image
draw = ImageDraw.Draw(HBlackimage)
drawry = ImageDraw.Draw(HRYimage)

drawry.rectangle((0,0,epd.height,epd.width), fill = 0)
draw.text((0, 0), name, fill = 1, font=intuitive_font)
epd.display(epd.getbuffer(HBlackimage), epd.getbuffer(HRYimage))
```

SSYYL, 2020-09-16:
try ：
draw.text((0, 0), name, fill = 1, font=intuitive_font)
-》
draw.text((0, 0), name, fill = 0, font=intuitive_font)

hakarune, 2020-09-17:
> try ：
> draw.text((0, 0), name, fill = 1, font=intuitive_font)
> -》
> draw.text((0, 0), name, fill = 0, font=intuitive_font)

I tried this and it did not work. I looked more into it and spent over an hour reading through code and testng. No matter what I tried, black cannot write over red. Red can write over black, but not the other way around. I was able to get white over black, red over black and white over red, but black over red did not work no matter what I tried....

SSYYL, 2020-09-17:
Red and Blakck/White is mutually exclusive.
[image]

SSYYL, 2020-09-17:
draw a black rectangle and draw red text.
And use reversal display, like this:
```
self.send_data(~blackimage[i])     
self.send_data(~ryimage[i])   
```

SSYYL, 2020-09-17:
```
drawry.rectangle((0,0,epd.height,epd.width), fill = 0)
drawry.text((0, 0), name, fill = 1, font=intuitive_font)
draw.text((0, 0), name, fill = 0, font=intuitive_font)
```

hakarune, 2020-09-18:
> draw a black rectangle and draw red text.
> And use reversal display, like this:
> 
> ```
> self.send_data(~blackimage[i])     
> self.send_data(~ryimage[i])   
> ```

Where would I put this?

SSYYL, 2020-09-18:
> > draw a black rectangle and draw red text.
> > And use reversal display, like this:
> > ```
> > self.send_data(~blackimage[i])     
> > self.send_data(~ryimage[i])   
> > ```
> 
> Where would I put this?

It is not good, you can try
> ```
> drawry.rectangle((0,0,epd.height,epd.width), fill = 0)
> drawry.text((0, 0), name, fill = 1, font=intuitive_font)
> draw.text((0, 0), name, fill = 0, font=intuitive_font)
> ```

hakarune, 2020-09-19:
> t is not good, you can try

Yeah that is what I ended up doing for the text and some shapes. tis a shame there isn't a way to reverse which one writes on top. Thank you for your help.

SSYYL, 2020-09-19:
logging.info("1.Drawing on the Horizontal image...") 
    HBlackimage = Image.new('1', (epd.height, epd.width), 255)  # 298*126
    HRYimage = Image.new('1', (epd.height, epd.width), 255)  # 298*126  ryimage: red or yellow image  
    drawblack = ImageDraw.Draw(HBlackimage)
    drawry = ImageDraw.Draw(HRYimage)
    drawry.rectangle((0,0,epd.height,epd.width), fill = 0)
    drawry.text((0, 0), 'hello', font=font24, fill = 1)
    drawblack.text((0, 0), 'hello', font=font24, fill = 0)
    epd.display(epd.getbuffer(HBlackimage), epd.getbuffer(HRYimage))
    time.sleep(2)
I tried it. It works
[image]

closed 2020-12-04 by SSYYL