waveshareteam/e-Paper #99: python test crashes rasberry pi zero w running raspbian lite

ethugbaby, 2020-09-19:
```
pi@raspberrypi:~/e-Paper/RaspberryPi&JetsonNano/python/examples $ sudo python3 epd_2in13_V2_test.py
INFO:root:epd2in13_V2 Demo
INFO:root:init and Clear
```
when i run the example it starts to load nothing happens on the epaper screen and then the device crashes and reboots the raspberry pi 0 w

missionfloyd, 2020-09-20:
The HAT doesn't work right with the Pi 0. You have to use the cable.

coredump2020, 2020-09-20:
@ethugbaby you can take a look at #92 .

Kreshnik, 2020-10-17:
> The HAT doesn't work right with the Pi 0. You have to use the cable.
> 
> They know about the problem and say they may fix it in the future.

The cable doesn’t work either

missionfloyd, 2020-10-18:
Are you connecting VCC to 3.3V? Connecting it to 5V is what makes it crash.

Kreshnik, 2020-10-18:
I've recently received the 7.5 inch display, tried it with Raspberry Pi 4/zero, I can't get it even to work. 
I noticed that my display doesn't look the same as everybody's else's, I think I received a different display....  [Screen image](https://imgur.com/a/tDLdHxD)

missionfloyd, 2020-10-19:
That's the 880x528 black/white/red panel. Try epd_7in5b_HD_test.py

Kreshnik, 2020-10-19:
> That's the 880x528 black/white/red panel. Have you tried [epd_7in5b_HD_test.py](https://github.com/waveshare/e-Paper/blob/master/RaspberryPi%26JetsonNano/python/examples/epd_7in5b_HD_test.py)?

Yes, it worked also I had  the interface wrongly attached. Thank you @missionfloyd