waveshareteam/e-Paper #100: 7in8 support missing?

vanarebane, 2020-09-24:
Currently there is no working sample code for 7.8" display

Even the [C sample here](https://www.waveshare.com/wiki/7.8inch_e-Paper_HAT) is just a lazy copy from 6" display

dperkins0, 2020-10-08:
Hey @vanarebane , after a few days of messing around trying to get my 7.8" display working I found a few things -
1) better documentation - https://www.waveshare.com/wiki/6inch_HD_e-Paper_HAT
2) Some pins were upside down. I had to turn my ePaper adapter upside down. Check out the pictures in the wiki article above.
3) working code! have a look at this repo over here - https://github.com/waveshare/IT8951-ePaper

After cloning that repo, I was able to run `sudo ./epd -1.36` and success!

GammaGames, 2020-11-05:
Thank you for the help @dperkins0, I had followed the images on the wiki for the 7.8 display and I had the FPC cable upside down. Looks like my display is -1.34 too, not sure why there is a variation. 

Now to try and write a python wrapper...

dperkins0, 2020-11-05:
After a lot of frustration I also worked out that the cable was upside down.

As for a python wrapper - https://github.com/GregDMeyer/IT8951

briancwald, 2020-11-09:
@dperkins0 what was the error that made you determine the cable was upside down? I am pretty sure I have it connected correctly but when I run `make` in the `IT8951-ePaper` folder I am running into this error:

`In file included from lib/Config/DEV_Config.h:60,
                 from lib/Config/DEV_Config.c:30:
/usr/local/include/bcm2835.h:643:8: error: unknown type name ‘off_t’
 extern off_t bcm2835_peripherals_base;`

I am trying to track down what that is referring to.

dperkins0, 2020-11-11:
@briancwald I didn't get any errors out - just no display on the screen. There should be pin numbering on the driver board where the ribbon cable connects, as well as numbering on the connector at the screen end. I just made sure those numbers lined up.

mweschan, 2020-12-17:
@briancwald I'm getting the same error that you got at that step. Were you able to find a solution?

briancwald, 2020-12-17:
@mweschan - no I ended up abandoning that and using this library https://github.com/waveshare/IT8951 instead, which I managed to get work.

mweschan, 2020-12-17:
@briancwald Thank you for the advice, I'll go in that direction.

SSYYL, 2021-05-06:
Here is the new driver that supports all WaveShare parallel port screens. [IT8951 driver](https://github.com/waveshare/IT8951-ePaper)

closed 2021-05-06 by SSYYL

js-compilatrum, 2024-05-21:
Where can I find example for this model:

https://www.waveshare.com/wiki/7.8inch_e-Paper_HAT#Demo_Description

At directory is not available? Is it not supported?