waveshareteam/e-Paper #105: Array overflow in GUI code.

rohoog, 2020-10-31:
When compiling with optimization (change -O0 into -O in CFLAGS) I noticed segmentation fault at example program completion.
I tracked it down to a return-address corruption, caused by array overflow in GUI_BMPfile.c (line 96).

The patch to fix this:

diff --git a/RaspberryPi&JetsonNano/c/lib/GUI/GUI_BMPfile.c b/RaspberryPi&JetsonNano/c/lib/GUI/GUI_BMPfile.c
index 707f97a..bd4c134 100644
--- a/RaspberryPi&JetsonNano/c/lib/GUI/GUI_BMPfile.c
+++ b/RaspberryPi&JetsonNano/c/lib/GUI/GUI_BMPfile.c
@@ -93,7 +93,7 @@ UBYTE GUI_ReadBmp(const char *path, UWORD Xstart, UWORD Ystart)
 
     for(i = 0; i < bmprgbquadsize; i++){
     // for(i = 0; i < 2; i++) {
-        fread(&bmprgbquad[i * 4], sizeof(BMPRGBQUAD), 1, fp);
+        fread(&bmprgbquad[i], sizeof(BMPRGBQUAD), 1, fp);
     }
     if(bmprgbquad[0].rgbBlue == 0xff && bmprgbquad[0].rgbGreen == 0xff && bmprgbquad[0].rgbRed == 0xff) {
         Bcolor = BLACK;

SSYYL, 2020-12-04:
Yes, it is.

SSYYL, 2020-12-14:
This bug has been fixed, Thanks.

closed 2020-12-14 by SSYYL