waveshareteam/e-Paper #111: Could not determine Jetson model (Python)

TonySpegel, 2020-11-11:
I followed the instructions as stated [here ](https://github.com/waveshare/e-Paper/blob/master/RaspberryPi%26JetsonNano/python/readme_jetson_EN.txt) and while running one of the examples only got 
`Could not determine Jetson model`

Any ideas?

SSYYL, 2020-12-29:
Maybe the kernel update made the identification method invalid, which we will check later.

mikechase3, 2021-01-16:
I'd be interested in an answer too if there are any updates.

mikechase3, 2021-10-19:
> Any updates on this? Im having the same issue

I got way too stuck and gave up. Please let us know if you figure it out though!

TheNukeLore, 2022-05-31:
anyone have any updates on this? Im having a similar issue and am having an extremely hard time trying to get such working.

mikechase3, 2022-05-31:
> Hi,
>
> I just did a fresh install of raspian and it started working. Hope this helps
>
> BR,
>
> Shareef

Never got it working and gave-up after ~5hrs. Good luck!

stubby-phillips, 2023-01-24:
Worked for me:
sudo pip3 uninstall Jetson.GPIO
Found existing installation: Jetson.GPIO 2.0.20
Uninstalling Jetson.GPIO-2.0.20:
  Would remove:
    /usr/local/lib/python3.9/dist-packages/Jetson.GPIO-2.0.20.dist-info/*
    /usr/local/lib/python3.9/dist-packages/Jetson/*
    /usr/local/lib/python3.9/dist-packages/RPi/*
Proceed (y/n)? y
  Successfully uninstalled Jetson.GPIO-2.0.20

$ python3 ./main.py

ID Read success  148088885
0 ADC = 0.071160
1 ADC = 0.071474
2 ADC = 0.071619
3 ADC = 0.070657
4 ADC = 0.071288
5 ADC = 0.071562
6 ADC = 0.071720
7 ADC = 0.070421
DAC : 0.07095445181963836