waveshareteam/e-Paper #113: Raspberry Pi Zero W crash during Init

romoromor, 2020-11-30:
Hi,

we have Raspberry Pi Zero W ad into this connected epd2in13bc (Waveshare® 2.13 Inch E-ink Screen Display e-Paper Module SPI Interface Partial Refresh Black Red White 212×104).
When we execute script to view something on display for example in a hour after start of Raspberry, all is ok.
But when we leave Raspberry running and do script execution after more than +- 3hours, Raspberry crashed during sending first command to display (without any error message).
**epd.init()**
Please, what could be wrong?
We have latest version of e-Paper, updated Raspberry, Raspberry is powered by power supply (not from PC USB).

SSYYL, 2020-12-03:
Does it come up every time?  When did you buy your HAT?

romoromor, 2020-12-03:
Hi, if i set refresh once per day at 12, everyday in that time is Raspberry hard restarted.
syslog etc. are empty from that time. No message about shutdown, only starting of OS  messages.
I bought HAT and Raspberry second week of October this year.
Here is screen from monitoring. You can see uptime there. 28.11.2020 was crontab task disabled and in graph are also my two reboots of device - 25.11. 8AM and 27.11.. 3PM.
[image]

LoieBensbane, 2020-12-03:
There is a known problem (#99 ) with connecting the HATs to pi Zero (the module+display may pull 5V, overvolting the Pi). There is no software solution I know of. A workaround is to use a ribbon cable to attach the HAT to Pi.

Double check your pin connections if you use the cable. VCC should be on 3.3V.

I hope that helps.

SSYYL, 2020-12-04:
Indeed, the VCC connection to 5V is one of the causes of Pi's crash.
If HAT is plugged directly into PI ZERO, the Vcc is connecting to 3.3V.

So i think this is a hardware (Duplicate of 92#)

SSYYL, 2020-12-04:
Indeed, the VCC connection to 5V is one of the causes of Pi's crash.
If HAT is plugged directly into PI ZERO, the Vcc is connecting to 3.3V.

So i think this is a hardware (Duplicate of #92 )

SSYYL, 2020-12-04:
If your PI crashes every time at reset (), contact the after-sale service to reissue a new version HAT.

romoromor, 2020-12-04:
Thanks for answer.
I did connection with cables and first try was successfull,
I leave it for a weekend and if whole weekend will be Raspberyy UP, we can close this issue 👍 
Do you know, if will exist some upfgraded revision of HAT hardware without this issue. Or once solution is soldering od 3,3V to 5V pin?

LoieBensbane, 2020-12-04:
Good to hear it's working for you now!

The issue that @SSYYL mentioned (#92 ) has some suggestions for modifying the hardware if you scroll past the "closed this issue" message. I only recognized some of your symptoms from reading other setup issues with Pi Zero. I don't know what I'm doing when it comes to soldering, so if this issue was mine, I would contact Waveshare to see if they can assist more with repair or replacement.

I had meant to order a epd2in13bc HAT (C) for use with my Pi Zero WH, but the smaller HAT was the same price as the larger epd2in9bc module (C) and there was no "action shot" of the screen connected to something for reference size. Shipping took months to get to me, so returning it was not desirable. I wasn't concerned with the pinout until I opened the box and it wasn't a HAT. I had to look at pinout maps for a couple days and read half a dozen tutorials before I felt comfortable plugging the screen into my Pi at all.

SSYYL, 2020-12-07:
@romoromor I'm glad your Pi is working. It is a pity that there is no new version at present, But it's already in the plans.

romoromor, 2020-12-08:
Hi, super, display still working without reset. I think, that is a workaround for me:-)
Thanks, you can close this case.

closed 2020-12-09 by SSYYL