waveshareteam/e-Paper #115: Support for EPD_7in5bc_V2 missing

dbreunig, 2020-12-10:
The file described [here](https://www.waveshare.com/wiki/7.5inch_HD_e-Paper_HAT_(B)) to support the V2 7.5 inch red and black screen does not appear to be present any longer. Can we add it back to the library or offer an alternative?

SSYYL, 2020-12-14:
Update 7.5inch_e_Paper_HAT(B) instead of 7.5inch_HD_e_Paper__HAT(B).
look here: [https://www.waveshare.com/wiki/7.5inch_e-Paper_HAT_(B)](url)
The code: [https://github.com/waveshare/e-Paper/blob/master/RaspberryPi_JetsonNano/python/lib/waveshare_epd/epd7in5b_V2.py](url)

SSYYL, 2020-12-14:
Update 7.5inch_e_Paper_HAT(B) instead of 7.5inch_HD_e_Paper__HAT(B).
look here: [7.5inch e-Paper(B)](https://www.waveshare.com/wiki/7.5inch_e-Paper_HAT_(B))
The code: [7in5b_V2](https://github.com/waveshare/e-Paper/blob/master/RaspberryPi_JetsonNano/python/lib/waveshare_epd/epd7in5b_V2.py)

dbreunig, 2020-12-15:
That doesn't work. Code hangs without updating.

epd7in5b_V3 works great on RPi3 and Zero.

SSYYL, 2020-12-15:
Hi, I'm not sure what you mean. Please describe your problem in detail.
7.5b_V2 Code is support version V2 and V3.

dbreunig, 2021-02-01:
Hi, in the current release, the python demo code (7in5b_V2) and my own code perpetually hangs on e-Paper Busy. I am having a hard time debugging this and can't get more information.