waveshareteam/e-Paper #116: Missing partial refresh in the source

denisjoshua, 2020-12-14:
Hi there, 
I just buy a 2.13 e-paper wit black and red color. 
But in the driver (epd2in13bc) I didn't find the partial refresh method. 
So how can I use the partial refresh on this e-paper please ? can you update the driver ? 
Thanks in advance

denisjoshua, 2020-12-14:
p.s. can't fined lut_full_update and lut_partial_update

SSYYL, 2020-12-14:
Hey @denisjoshua , I'm sorry to tell you that there is no partial refresh on the three-color screen.

denisjoshua, 2020-12-14:
Thanks for the answer... but can you send me a link please of a e-paper that have a pertial refresh please (from 2 to 3 inch) 
Maybe a product that allready have EPS32 on board
Thanks a lot again

SSYYL, 2020-12-14:
2.13inch (BW) have partial refresh. You can look here: [ESP 32 e-Paper Board](https://www.waveshare.com/wiki/E-Paper_ESP32_Driver_Board) and here: [2.13inch e-Paper](https://www.waveshare.com/wiki/2.13inch_e-Paper_HAT)

denisjoshua, 2020-12-14:
Thanks a lot :-)

closed 2020-12-19 by SSYYL