waveshareteam/e-Paper #117: 3.7" supports partial update, but has no partial update code in python driver

eugenefischer, 2020-12-15:
According to the [product page](https://www.waveshare.com/3.7inch-e-paper-hat.htm), the 3.7" e-paper display supports partial update, but there doesn't seem to be any code in the python driver to use it. The only display methods are for the full display.

SSYYL, 2020-12-17:
Hi, The "partial refresh" mean there is no flicker when updating. 
If you want to get a faster refresh rate.
Please open epd3in7.py
in `def display_1Gray(self, image):`
change `self.load_lut(self.lut_1Gray_DU)` to `self.load_lut(self.lut_1Gray_A2)`

If you want to do part of the data transfer, You can do it according to the c , and if there is a difficulty we will update later.

About why it's not recommended, because it's really hard to use.
I said that in the notes:
    1.Xsize must be as big as EPD_3IN7_WIDTH
    2.Ypointer must be start at 0

TomBric, 2020-12-23:
Hi, I am experiencing the same issue with refresh rates on the 3.7 ePaper. The specs said the partial refresh rate will be shorter than a second, but with the libraries and examples provided it takes on a Pi Zero approx. 4 seconds. In this way I can not use the display for the intended application.
Also applying the recommended changes in display_1Gray does not really help. It is the transfer (send_data) that consumes the most significant portion of time. I there any way to speed up the send_data functions significantly?

SSYYL, 2020-12-25:
Hi @TomBric , The partial refresh time refers to the time it takes to refresh without the data being transmitted, Because different devices have different transmission times that would be unfair.
If you want to speed up send_data funtions:
1.Increase the SPI rate.
2.Use lists to send multiple data at once rather than byte by byte.(We use bytes for ease of learning and stability)

TomBric, 2020-12-26:
Hi SSYYL,
using lists to send the image really is the solution. I am down to a refresh time of 0.8 secs now on a PI3B+ and on Pi Zero W at 1.2 secs. You should include this in your demo code ...

eugenefischer, 2020-12-26:
I'll have to try using lists as well. Thank you for testing it, TomBric. I, too, would really like to see this included in the demo code. TomBric, would you happen to be willing to share your own testing code sending image data with lists?

TomBric, 2020-12-26:
Sure, I can provide this. I will upload my modified code to my repo. Give me one day ...
BTW: I also implemented asynchronous waiting for the display to finish. This way other coroutines of the program can continue.

eugenefischer, 2020-12-27:
Awesome, look forward to seeing it. Thank you.

TomBric, 2020-12-27:
@eugenefischer: Repo is available:   https://github.com/TomBric/stratux-epaper-radar.
The relevant parts for you are in epdconf.py and epd3in7.py files. 
I also set the SPI speed to the max value that worked reliable (32 MHz).

eugenefischer, 2020-12-27:
This is tremendously helpful, thank you. Now I can get the 3.7" display to refresh fast enough for real-time typing

closed 2021-02-20 by SSYYL