waveshareteam/e-Paper #118: Possible command overwrite on epd_2in13_V2_test.py

bestbug456, 2020-12-20:
Hello,

I've ported your driver in epd_2in13_V2_test.py to Golang but the porting did not work until I start calling the readyBusy function inside the send_command and send_data function. For what I was able to understand the stack python was slow enough to avoid the override, but the problem should be present in the python driver as well. Beside the problem I've found I still suggest to add defensive check in order to reduce to zero the possibility to overwrite.

SSYYL, 2020-12-25:
Thanks, we will add defensiv check. If you can, you can give specific examples of the problems
By the way, our code is mainly for ease of learning and stability, so we don't do too much optimization.

bestbug456, 2021-01-02:
Hello and happy new year!

For what I was able to understand in the python library there are no check when multiple bytes are send to e-paper, thus if you write faster than epaper can read you lose one or more byte. I was not able to create this problem on python probably, as I was referring in my original post, this is caused by the fact python is an interpreted language where Golang is a compiled language. Hope this help you to understand the problem I've found. Have a great day

solvaholic, 2021-01-02:
> ... the porting did not work until I start calling the readyBusy function inside the send_command and send_data function.

I think these are the `send_command` and `send_data` functions @bestbug456 means

https://github.com/waveshare/e-Paper/blob/73946511e07b83ccabcb7f848d11861e4be86339/RaspberryPi_JetsonNano/python/lib/waveshare_epd/epd2in13_V2.py#L95-L105

And this the `ReadBusy` function:

https://github.com/waveshare/e-Paper/blob/73946511e07b83ccabcb7f848d11861e4be86339/RaspberryPi_JetsonNano/python/lib/waveshare_epd/epd2in13_V2.py#L107-L109

I think it looks like calling `ReadBusy` will block other commands' execution until `digit_read` returns `0`, for idle.

Because `send_command` and `send_data` do NOT block after sending commands or data, it's possible a subsequent command could interrupt and replace a command that's already running to the display.

I'm unsure what problems that would cause. I suspect it would look, to the user, like the display had failed to complete a refresh.

solvaholic, 2021-01-02:
> Because `send_command` and `send_data` do NOT block after sending commands or data, it's possible a subsequent command could interrupt and replace a command that's already running to the display.

Looking closer at [epd2in13_V2.py](https://github.com/waveshare/e-Paper/blob/66ff76f87ecac4f7ef338347695481ca14b58db1/RaspberryPi_JetsonNano/python/lib/waveshare_epd/epd2in13_V2.py), when other functions call `send_command` or `send_data`, they usually also call `ReadWait` after.

It would help to know which function call was impacted. @bestbug456 do you happen to recall which function failed?

bestbug456, 2021-01-03:
Take for instance this set of commands

[Initialisation](https://github.com/waveshare/e-Paper/blob/66ff76f87ecac4f7ef338347695481ca14b58db1/RaspberryPi_JetsonNano/python/lib/waveshare_epd/epd2in13_V2.py#L165)
```
            self.send_command(0x04) #
            self.send_data(self.lut_full_update[71])
            self.send_data(self.lut_full_update[72])
            self.send_data(self.lut_full_update[73])
```

Here the driver is sending multiple data byte without verify the e-paper is ready for receive the next one. Beside this even this sample

```
            self.send_data(0x1)
            self.send_data(0x2)
            self.send_data(0x3)
```

Could potentially result in an override of the data the e-paper should read since between the send_data no check is done. As I was previously say I was not able to reproduce this problem in python but only on Go. For instance in my test when I try to create the image and using send_data and send_command "a lot" I was able to see the override.

SSYYL, 2021-01-05:
happy new year!

Adding too much CheckBusy can cause the refresh to take too long.(Many people complain about this)
Normal data and commands don't cause 'busy', just need to add CheckBusy in the key places.

bestbug456, 2021-01-05:
Ok if you have already investigated this situation probably the problem is related to the execution enviroment, in Go I was not able to see any slowdown in increasing the number of busy (beside the fact that I was not able to have a correct update in GO without extensive checkBusy). At this point if the code from your side is working as intended I think we can close the issue.

SSYYL, 2021-01-06:
Yes, it is fine in our testing environment and we will strengthen testing in the future.

Thank you for your feedback.

bestbug456, 2021-01-06:
Have a nice day then! :+1:

closed 2021-01-06 by bestbug456