waveshareteam/e-Paper #124: Disable Flickering?

OleAlbers, 2020-12-29:
I use the simple task of displaying a two-colored - image:
(code is very much copy/paste from example)
```
epd=epd7in5b_HD.EPD()
epd.init()
blackimage = Image.open(os.path.join(picdir, "black.bmp"))
redimage = Image.open(os.path.join(picdir, "red.bmp"))
logging.info("start")
epd.display(epd.getbuffer(blackimage),epd.getbuffer(redimage))
```

While this DOES work I have the issue that the display is flickering very fast for about 15 seconds before the image is displayed. Is there any way to get around this issue?

OleAlbers, 2020-12-29:
closed because it is a hardware limitation

closed 2020-12-29 by OleAlbers

gaul, 2020-12-30:
Could someone explain the hardware limitation?  It is interesting watching the flickering and I wonder what is going on?

OleAlbers, 2020-12-30:
Not why it is flickering, but on the WaveShare "Selection-Guide" (end of page) there is a list which display needs how many seconds to do this ("Total Refresh Time")

In Addition SOME display allow partial refresh which should reduce flickering

https://www.waveshare.com/product/displays/e-paper/epaper-1/12.48inch-e-paper-module-b.htm